UltimateHackingKeyboard/firmware #1035: Issue with the firmware 12.0.0 and trackball module

Jopie01, 2024-12-16:
Today I upgraded the firmware and the agent to the latest versions:
- firmware: 12.0.0
- agent: 5.0.0

It seems there is also a new configuration version so I migrated to that as well. But now I walk into trouble with my trackball.

I have mapped the left button of the trackball on layer MOD as middle mouse click. See the screenshot from version 4.2.1
[image]

I did the same with the new configuration version, new firmware and agent, but cannot get it working.

kareltucek, 2024-12-16:
Could you please export your UserConfiguration.json in the Configuration tab? You can rename it to .txt so that github accepts it...

kareltucek, 2024-12-16:
Although, nevermind, this config looks pretty standard. Will try to test with a factory settings.

Jopie01, 2024-12-16:
> Although, nevermind, this config looks pretty standard. Will try to test with a factory settings.

Yeah, I don't know what the `fix configuration` button in the agent does, but it seems it just sets a factory one. The only thing I changes was adding the middle click and lowered the led brightness.

Totoro83, 2024-12-17:
Not sure if additional feedback is something you'd like to have, as I see that there is already a pull request that should solve the issue, but I have exactly the same issue (except that I mapped the buttons "backward" and "forward" in the mod layer.)

closed 2024-12-18 by kareltucek

mondalaci, 2024-12-18:
Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12392454444 release resolve your issues?

Jopie01, 2024-12-27:
A very late response, but I want to let you know that the newest firmware 12.1.0 fixes this issue.

Totoro83, 2024-12-27:
It does for me too.

PR #1036 (2024-12-18): Fix load of default module keyActions.
Closes #1035