UltimateHackingKeyboard/firmware #1039: Agent 5.0, Firmware 12.0.0 - weird mouse wheel behaviour

abonhote, 2024-12-17:
Hi all

Since I updated to 12.0.0 today, I'm experiencing a weird mouse wheel behaviour, at least while using Windows 10. Not all applications are affected. The developer console of the UHK agent seems to work fine, while my SQL tool of choice (allroundautomations.com => PL/SQL developer) refuses to scroll. Also notepad++ has problems scrolling, it needs two keystrokes to scroll down.

I'd love to give you more information, but I'm not sure how. Please tell me what you need to investigate on it.

Cheers

André

mondalaci, 2024-12-17:
I agree that hi-res scrolling is probably the culprit since 12.0.0 is the first firmware release with #794 merged.

Related issue: #741

I can reproduce the issue in Notepad++ on Windows 11.

@rightaditya @mhantsch Any insights?

abonhote, 2024-12-17:
Is there a way to turn this off?

mondalaci, 2024-12-17:
There's no way to turn this off. Downgrading Agent and the firmware is the only way.

rightaditya, 2024-12-17:
Hmm, OK, it won't be so simple to revert *just* for the mouse keys since they still need to send values scaled according to the OS's (new) expectation.

Since I can't test it myself I might as well describe what I think is going on... The keyboard reports high-res support and so the OS expects 120x the scroll values from the mouse device. The mouse keys apply the multiplier when scrolling, but the mouse controller sends whatever events it has accumulated every time it sends its report, without waiting for to accumulate a full detent's worth (this is required for the high-res scrolling to work). The OS receives what would ordinarily be a "partial" scroll (i.e., not a full detent) but some applications explicitly wait for a full detent, and so need more than one press of the scroll key (or holding it down) to actually scroll.

So this might be addressable by adjusting scroll speed settings for the mouse keys. It sounds like Notepad++ might be waiting for a full scroll detent, since it responds to two presses of the scroll key. Doubling the scroll speed values might do the trick.

If so, an extra multiplier can be applied on top of the high-res scrolling multiplier in the firmware so that users' existing settings continue to work more-or-less as-is. The only potential issue is that the actual scroll amount sent from a scroll key is dependent on how long it's held down, so one user's "single press" is likely to be different than another's (or even their own).

If someone can confirm that doubling (or applying some other adjustment to) the scroll speed values fixes it, I can try and find an appropriate multiplier value to use and write the patch.

mondalaci, 2024-12-17:
Thanks for your quick follow-up and insight!

My settings:

[image]

I don't notice the issue when using my trackpoint module, but it's easily reproducible with mouse keys, especially when switching scrolling directions and scrolling momentarily.

When I increased the initial scrolling speed to 61, I couldn't notice the issue with the mouse keys anymore.

mhantsch, 2024-12-17:
I would also argue that instead of attempting quick fixes with firmware updates, users with this particular problem should look at their mouse scroll speed settings in Agent. I know that @rightaditya made sure the same multiplier is applied with key presses that activate mouse scrolls because that was an issue I had in Windows and his changes corrected that.

I suspect I just use a higher base scroll speed value and that's why I don't get the "two key presses needed" before a scroll happens.

I have to check my settings but I am not at the UHKs right now.

abonhote, 2024-12-18:
> @abonhote how are you triggering the mouse wheel? Using a module (Touchpad, trackball, trackpoint, mini-trackball) or from key presses?
> 
> What are your mouse-relevant settings in Agent?

[image]

Interesting Axis skew value ...

abonhote, 2024-12-18:
I changed the initial scrolling speed to 63 and the axis skew to 1. Now scrolling downwards works, but upwards doesn't 😉

[image]

I'm such a bug magnet ... damn ...

abonhote, 2024-12-18:
> There's no way to turn this off. Downgrading Agent and the firmware is the only way.

Just out of curiosity: did you try that? I downloaded the last agent installer, it says it couldn't find my UHK.

abonhote, 2024-12-18:
> There's no way to turn this off. Downgrading Agent and the firmware is the only way.

I just learned that the agent updates itself after each start ... I would love to have an option to turn this behaviour off ...

pcooke9, 2024-12-18:
> I just learned that the agent updates itself after each start ... I would love to have an option to turn this behaviour off. Now I have to downgrade after each start.

Temporarily disable your internet before downgrading the next time and then look here after installing:

[image]

abonhote, 2024-12-18:
> > I just learned that the agent updates itself after each start ... I would love to have an option to turn this behaviour off. Now I have to downgrade after each start.
> 
> Temporarily disable your internet before downgrading the next time and then look here after installing:
> 

Well, "check for update" is not necessarily "install update", right? It's a bit misleading IMHO.

pcooke9, 2024-12-18:
> Well, \"check for update\" is not necessarily \"install update\", right?

I don't disagree; it would probably be better to notify of an update and require confirmation to install.

kareltucek, 2024-12-18:
Why would you check if you didnt want to install though?

mhantsch, 2024-12-18:
Yeah, I just checked, I use much higher pulses/s for the Mouse scroll speed, and that gives me no issues with scrolling from key presses:

[image]

mhantsch, 2024-12-18:
@abonhote What's your experience with scrolling if you use my "Mouse scroll speed" settings?

abonhote, 2024-12-18:
> Why would you check if you didnt want to install though?

Best use case is the current situation. I'd like to get notified when a new version is around, but I won't install it unless this issue here is resolved.

abonhote, 2024-12-18:
> @abonhote What's your experience with scrolling if you use my "Mouse scroll speed" settings? They work well for me no matter whether I use the mini-trackball from the key cluster, or scroll keys (mouse scroll actions assigned to key presses), or I use scrolling with the touchpad or trackball modules.

I downgraded to 11.2.0. Your settings work fine with this fw. Don't think I will flash 12.0.0 before the next update.

pcooke9, 2024-12-18:
> Why would you check if you didnt want to install though?

I'd prefer to read the changelog before updating. And in cases like we see here, it might also be better to leave the auto update option unchecked for first time/clean installations.

mhantsch, 2024-12-18:
> Why would you check if you didnt want to install though

Just to see if there is something new, and study the release notes before deciding to upgrade.

mhantsch, 2024-12-18:
> I downgraded to 11.2.0. Your settings work fine with this fw. Don't think I will flash 12.0.0 before the next update.

These settings should work fine with 12.0.0 too. I've tested them with a special version of 11.2.0 that included the hires-scrolling patches -- the same code that is now also in 12.0.0.

mondalaci, 2024-12-18:
Please let's discuss update-related suggestions in the newly created https://github.com/UltimateHackingKeyboard/agent/issues/2465 issue, guys.

rightaditya, 2024-12-19:
> I downgraded to 11.2.0. Your settings work fine with this fw. Don't think I will flash 12.0.0 before the next update.

@abonhote 11.2.0 didn't have the patch that caused the issue. You'd need to try the different settings on 12.0.0+ or a [patched 11.2.0](https://github.com/rightaditya/uhkfirmware/releases/download/highres-2024-10-15/uhk-firmware-11.2.0-highres-2024-10-15.tar.gz) in order to confirm that the problem's fixed for your applications. From the discussion above it sounds like it should be at least for Notepad++, so hopefully should be for the other as well.

anb-apg, 2024-12-27:
@rightaditya @mhantsch @mondalaci I upgraded to 12.1.0 today and adapted the settings to the ones provided by @mhantsch. Scrolling works fine in most apps, but not in all. We have a business ERP written in Java Swing. Scrolling is sluggish, even when accellerated, and zooming (using CTRL) doesn't work at all using the keyboard. Both works fine with the trackpad and the trackball.

So I guess there's still some work to be done here, I'm afraid. I'm happy to test whatever beta you provide.

abonhote, 2024-12-27:
@rightaditya @mhantsch @mondalaci I upgraded to 12.1.0 today and adapted the settings to the ones provided by @mhantsch. Scrolling works fine in most apps, but not in all. We have a business ERP written in Java Swing. Scrolling is sluggish, even when accellerated, and zooming (using CTRL) doesn't work at all using the keyboard. Both works fine with the trackpad and the trackball.

So I guess there's still some work to be done here, I'm afraid. I'm happy to test whatever beta you provide.

(edit: posted this from the wrong account)

mondalaci, 2024-12-28:
Some of these issues may have been fixed by the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12509029737 build.

abonhote, 2025-01-09:
> Some of these issues may have been fixed by the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12509029737 build.

Hi! Is this already in 12.2.0? I dug out an old mouse, just to check the behaviour. The scroll wheel of the mouse is much smoother than the scroll function on with the keys. As well, in our companies ERP application (java swing based), the keys behave irrationally, scroll back and forth. Zooming in on a map inside that ERP doesn't work at all (and I am responsible for that map component ... bummer ...)

Do you need anything else to analyze that thing?

pcooke9, 2025-01-10:
I think most of these issues might be fixed with: https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12706975201

You may need to adjust scroll speed settings for the modules. My touchpad scrolling was slow until I droped my scroll speed divisor.

abonhote, 2025-09-17:
> Is this ticket still relevant? If so, let me know and I will reopen it...

Hi Karel

Well, I got used to the behaviour, but it's still annoying. My expectation is that scrolling with the keyboard (Mouse modifier, Mouse->Scroll up/down action) behaves the same way as the scroll wheel of my mouse. If this is what you intend to achieve, then it's still relevant. Our Java Swing based ERP doesn't scroll at all, nor works zooming in the map component. Same with the SQL IDE I mentioned, it doesn't scroll. What I am also experiencing on a daily basis: beeps when I scroll in large word documents, sometimes also in other applications. It doesn't always happen, but it does - but not when I use the mouse wheel.

If you think the mouse scroll actions should act like the mouse wheel does, the ticket should be reopened.

Win10 user here ...

Cheers

André

mondalaci, 2025-09-17:
I suspect that the culprit is Windows 11, and more specifically, certain applications running on it. I'd rather round up scrolling offsets to full dents if partial dents cause the issue, and possibly only via a macro setting variable if this change would negatively affect the usability of mouse keys otherwise.

I'd wait for @rightaditya's answer regarding my question to gain more insight.

mondalaci, 2025-09-18:
@rightaditya Thank you for elaborating in such great detail! Is there a standard detent size (in scroll units), or is it OS-specific? If it is, do you think a UHK-configurable detent size would work as a workaround, so that the UHK would not output partial detents?

mhantsch, 2025-09-18:
@mondalaci I thought you need to output partial detents because that's what makes it scroll smoothly for apps that listen to hi-res scroll events. It was the OS driver (on the host) that accumulated small partial scrolls until it reaches 1 detent, then outputs a scroll event for apps that only handle low-res scroll events.

abonhote, 2025-09-18:
Wow, lots of traffic here. I'll read everything tomorrow, can't test today.

Thanks all!

mondalaci, 2025-09-18:
@abonhote After flashing the mentioned firmware build, please add the `set mouseKeys.scroll.forceFullFirstTick 1` macro command to the `$onInit` macro to test the fix and save your configuration.

[image]

abonhote, 2025-09-19:
> [@abonhote](https://github.com/abonhote) Could you check that the trackball still works if you move it much more slowly? And could you check if the mouse keys work if you bump up their speed to much higher values? Like 120+.

I actually never use the trackball to do anything. I tried to use it within one of the applications that has issues (plsqldev.exe). While scrolling up and down with the keyboard doesn't work, it DOES work with the Mouse modifier and the trackball! Same with zooming into the map in our ERP, works (CTRL-Mouse-scroll)

FW 15.0.1 on a UHK60, Win10

abonhote, 2025-09-19:
> [@abonhote](https://github.com/abonhote) After flashing the mentioned firmware build, please add the `set mouseKeys.scroll.forceFullFirstTick 1` macro command to the `$onInit` macro to test the fix and save your configuration.

Done. Didn't change anything, I'm afraid. Maybe this comment helps? https://github.com/UltimateHackingKeyboard/firmware/issues/1039#issuecomment-3310692480

mondalaci, 2025-09-19:
The latest build I referenced makes mouse keys emit a full dent when pressed initially. @kareltucek thinks that simulating full low-res scrolling wouldn't work either.

@rightaditya @benedekkupper Do you have any insights?

rightaditya, 2025-09-19:
> While scrolling up and down with the keyboard doesn't work, it DOES work with the Mouse modifier and the trackball!

@abonhote I may be missing something here... what do you mean by scrolling with the keyboard that wouldn't be with the mouse modifier?

abonhote, 2025-09-19:
> > While scrolling up and down with the keyboard doesn't work, it DOES work with the Mouse modifier and the trackball!
> 
> @abonhote I may be missing something here... what do you mean by scrolling with the keyboard that wouldn't be with the mouse modifier?

Keyboard means mouse modifier and keys bound to scroll up/down

abonhote, 2025-09-24:
All, thanks a lot for your help and your patience! I installed 15.2.0 #3b6eecb on my uhk60v2, set the param and I'm able to scroll in my SQL IDE 😊 Zooming in the Swing application also works again 👍

closed 2025-09-24 by abonhote