UltimateHackingKeyboard/firmware #1040: Agent/Firmware update right side issues

eletoblade, 2024-12-17:
I have a UHK 60 and just updated to agent 5.0.0 and firmware version 12.0.0. When I did so the right half of my UHK stopped working, all other parts (left, key cluster, trackball) of it work and connecting the halves does not solve my issue. When I revert my firmware to the previous version or create a new key map from a template the right half goes back to working.

kareltucek, 2024-12-17:
So the keyboard os dead only when it has 12.x firmware and your old config, right?

Have you tried to export the old config befor upgrade and then import it again after the upgrade?

eletoblade, 2024-12-17:
That is correct, I’m guessing it’s related to the new configuration options released with this version. I’m also not a a developer so 100% speculation on my end.

I have done so, unfortunately I still have the issue. What’s throwing me for a loop is the fact that it’s only the right half that stops working.

kareltucek, 2024-12-17:
So the problem is reproducible. Would you share that offending UserConfig? (You can rename it to a txt so that github accepts it as an attachment.)

eletoblade, 2024-12-17:
Let me know if this doesn’t work and I’ll add the body of the file in a comment 
[UserConfiguration.txt](https://github.com/user-attachments/files/18169552/UserConfiguration.txt)

kareltucek, 2024-12-18:
@eletoblade the fix for you is simple: open the json in some reasonable code editor and delete those surplus modules.

ert78gb, 2024-12-18:
If the user config is not modified by manually or by a 3rd party tool then can I get the original user config? You can find it in the user config history folder. There is the "Open configuration folder" link on the Settings page of the Agent. This link opens the Agent settings folder and in the `user-configs` sub folder contains the user configurations.
The file name structure is `yyyyMMdd-HHmmss-<hash>.bin` 
yyyy is the year
MM is the month
dd is the day
HH is the hour in 24 hours format
mm is minutes
ss is secconds
when the user config saved.

From Agent 5 we save the user config in device specific folder. So the files in the `user-configs` folder contains the user configs that saved by Agent 4 or earlier version

ert78gb, 2024-12-18:
I try to remote manage the debug process. Don't share any sensitive information.

Have you ever used any 3rd party tool that manipulated the config. I just would like to exclude the 3rd party tool case.

> create a new key map from a template the right half goes back to working.

How do you mean it exactly? You kept your user config and added a new empty keymap and modified it and everything started to work. Or did you use the reset user config function and customised it?

> what you mean by "surplus modules"?

Karel means, He attached a screenshot to the https://github.com/UltimateHackingKeyboard/firmware/issues/1040#issuecomment-2551317313 comment. 
You can see the `Layers` data structure has a `modules` structure. Every module has an id. A module could occur only once in the modules.

You can/must delete the last module with `0` of every `Layer` of every `Keymap`.

Agent 5 do some migration and I would like to investigate what cause this double `0` module situation. If it is a bug in our side I would like to fix it before it affects other users too.

What was your exact Agent 4 version before the Agent 5 upgrade?

eletoblade, 2024-12-18:
I've never used a 3rd party tool to edit my key maps, Agent is pretty good so I've never felt the need.

I created a new keymap from one of the template keymaps that comes with Agent (Dvorak for PC) and that one works fine, it's just all of my existing keymaps that have this issue.

Appreciate the clarity, I'll try that out.

I'll check on that and let you know. Is there a log of which agent version I was using? If so I'm guessing I can get to it through Agent itself or is there a log file somewhere I need to access?

I won't be in office (where I have my UHK) until tomorrow but I will post here with an update after I get into work.

eletoblade, 2024-12-19:
Unfortunately when I attempt to load that config into agent I gives me “invalid configuration specified”

it looks like agent version 4.11.0

ert78gb, 2024-12-19:
The Agent logs are available in the following folders.

on Linux: ~/.config/uhk-agent/uhk-agent.log
on macOS: ~/Library/Logs/uhk-agent/uhk-agent.log
on Windows: %USERPROFILE%\AppData\Roaming\uhk-agent/uhk-agent.log

The `invalid configuration specified` means Agent can parse the user config.
Did you try to load the new user configuration with the old agent or when did you get the error message? The old Agent can't parse the new user configuration. I checked the code and we don't sophisticate the reason.

eletoblade, 2024-12-19:
From that log file I’m guessing “deviceProtocolVerson” would be correct? If so it’s showing as 4.11

I tried loading the modified config with Agent 5.0.0

ert78gb, 2024-12-19:
`deviceProtocolVerson` is provided by the firmware. It describe which features supported by the firmware.
You can find the Agent version on the About page of the Agent. Anyway you are using Agent 5.0.0 otherwise it ask to upgrade to newer version.

> I tried loading the modified config with Agent 5.0.0

Do you mean you deleted the last `module` with `0 `of every `Layer` of every `Keymap`?
The user config file should be a valid JSON file. That means trailing comma is not supported. So when you deleted the `module` you also had to delete the previous comma too.
Could you double check not left any extra comma?

ert78gb, 2024-12-19:
BTW if you try to restore an archived user configuration will it fail? You could find the archived user configurations on the Configuration page of Agent

nathanph, 2024-12-20:
I was running into this same issue, deleting all of the empty modules[^1] seems to have fixed the issue.

I searched through the manual backups / exports of my UHK configs for that empty module pattern, but I couldn't pin it down to a particular agent version.

UHK backups with empty modules:
```
uhk-2024-01-12.json
uhk-2024-01-18.json
uhk-2024-03-10.json
uhk-2024-03-29.json
uhk-2024-04-11.json
uhk-2024-04-13.json
uhk-2024-04-17.json
uhk-2024-04-18.json
uhk-2024-04-23.json
uhk-2024-04-27.json
uhk-2024-04-28.json
uhk-2024-04-29.json
uhk-2024-10-15.json (last manual export I'd made before this most recent update - v7.1.0)
```

UHK files without empty modules:
```
uhk-2024-01-13.json
uhk-2024-01-14.json
uhk-2024-01-17.json
uhk-2024-12-20.json (this is uhk-2024-10-15.json with the empty modules deleted and exported - v8.3.0)
```

[^1]:
    Empty module example:
    ```json
    {
      "id": 0,
      "keyActions": []
    }
    ```

nathanph, 2024-12-20:
@eletoblade, in the meantime, grab an older version of Agent and "update" (revert) the firmware on your UHK to use your old configuration.

https://github.com/UltimateHackingKeyboard/agent/releases/tag/v4.2.1

ert78gb, 2024-12-20:
> Are they the .bin files in ~/.config/uhk-agent/user-configs?

yes they are the archived user-configs

> Can I convert those to JSON files somehow?

I could write a script, but currently not possible. Thinking on the solution

> in the meantime, grab an older version of Agent and "update" (revert) the firmware on your UHK to use your old configuration.

If you upgraded the firmware of the keyboard then Agent 4 for can't communicate with the new firmware.

@nathanph Could you share the version number of the user configuration where the extra modules with empty keyActions exist and not exists? 

Do you have backup before uhk-2024-01-12.json? If yes what is the user config version number and does it have empty actions?

eletoblade, 2024-12-20:
I double checked my commas, I missed some. Corrected my errors and now it’s working again 🎉

If I restore any of the archived configurations they don’t work either

ert78gb, 2024-12-21:
user config 6.0.0 officially contained

```
{
  "id": 2,
  "keyActions": []
}
```

but should not contain

```
{
  "id": 0,
  "keyActions": []
}
````

I have no clue how could it be possible.

I have extremely much archived user configuration and neither have duplicated `0` module and or `0` module with empty keymaps.
Until I can't reproduce the issue until I can't fix the original bug. Without the original bug fix I don't wanna put any extra step into Agent that fix this cases.

I would like to ask you to manually edit your user config or help me to reproduce the issue.

eletoblade, 2024-12-21:
I’m happy to help. I won’t be back in the office until 1/2 but what can I do to be helpful at that point? I’m just wondering how we would reproduce this. Maybe do a fresh install of a previous Agent on a new computer, do a config, and then update Agent?

ert78gb, 2024-12-22:
> Maybe do a fresh install of a previous Agent on a new computer, do a config, and then update Agent?

I did it yesterday. I can't reproduce the issue.

If I read well both of you recovered their user config by manually modification?

eletoblade, 2024-12-22:
Correct

kareltucek, 2025-09-16:
Closing since this has become stale.