UltimateHackingKeyboard/firmware #1041: Double tap with ifGesture does not work anymore

Totoro83, 2024-12-18:
Another regression with firmware 12.0.0.
The following macro is not working as intended anymore:

```
ifNotLayer fn holdLayer mod
ifLayer fn holdLayer mouse
ifNotLayer fn ifNotInterrupted ifGesture timeoutIn 300 $thisKeyId final toggleLayer mod
ifLayer fn ifNotInterrupted ifGesture timeoutIn 300 $thisKeyId final toggleLayer mouse
```

It should activate the layer "mod" or "mouse" respectively depending on whether "fn" is active or not (this works), and if I double-tap it, it should toggle the layer "mod" or "mouse" until another key is pressed. This last functionality seems completely broken. This macro always worked since I defined it (I don't remember when, but probably more than one year ago), and it stopped working correctly with this new firmware.

When I double-click the key, it sometimes work, sometimes don't, sometimes it works only if I tap it 3 times, and the corresponding "unToggleLayer" (mapped in the same key in the destination layer) don't work anymore.

I also noticed that I have a few errors logged in "agent", relative to this very macro:

```
Error at Mod (H) 1/1/1: 4198: Invalid abbreviation length: 101
> 1 | ifNotLayer fn holdLayer mod
Error at Mod (H) 1/1/1: 4188: 32 38 38 32 36 110 117 109 76 111
> 1 | ifNotLayer fn holdLayer mod
Error at Mod (H) 1/1/1: 4198: 99 107 79 110 32 38 38 32 33 36
> 1 | ifNotLayer fn holdLayer mod
Error at Mod (H) 1/1/1: 4208: 115 99 114 111 108 108 76 111 99 107
> 1 | ifNotLayer fn holdLayer mod
Error at Mod (H) 1/1/1: 4218: 79 110 41 32 115 101 116 76 101 100
> 1 | ifNotLayer fn holdLayer mod
Error at Mod (H) 1/1/1: 4228: 84 120 116 32 48 32 115 111 116 10
> 1 | ifNotLayer fn holdLayer mod
```

I'm not able to replicate them, and I'm not sure about when they where generated.

kareltucek, 2024-12-18:
Interesting.  

The last part is a config parser error/dump. Would you olease share your UserConfig?

Totoro83, 2024-12-18:
Sure. I'm sorry that's not a "minimal example".

The easiest way to test it is to try to doubletap on the "tab" button (the one that in the default configuration is configured as "mouse" button).

[UserConfiguration12.json](https://github.com/user-attachments/files/18188198/UserConfiguration12.json)

kareltucek, 2024-12-18:
No problem!

kareltucek, 2024-12-26:
@Totoro83 please, try to retest with 12.0.1 Agent. There was a toggle/untoggle regression which was very likely related. I haven't managed to reproduce the issue with current master.

As for those parser errors, they seem to be caused by the fact that the internal macro engine layer stack is not uninitialized before config save and re-initialized after. That is a known problem, but not a huge one.

Totoro83, 2024-12-27:
The firmware that "Agent" suggested me was the 12.1.0, I hope that that's the version you were talking about.
That said, yes, the problem seems to be solved with this new firmware. I've tested it for a bit and I wasn't able to find any problem. That said, I still had the parse errors when I restored my 11.2.0 configuration (taken from the configuration history) into the keyboard, but only at that point (and the keyboard seems to work fine after the restoring, even if I tested it only a few minutes) so I suppose that I can safely ignore the errors?

Thanks for the hard work, I really appreciate that the problem was solved fast despite the festivities 😄

closed 2024-12-27 by Totoro83

kareltucek, 2024-12-27:
> so I suppose that I can safely ignore the errors?

I believe so. But if there are any such troubles after powercycling the board, let me know!

Totoro83, 2024-12-27:
I don't think there is, I disconnected and reconnected the keyboard and everything seems right. I'll keep testing it, though. Thanks!