UltimateHackingKeyboard/firmware #1042: Reboot Zephyr after flashing the firmware

mondalaci, 2024-12-18:
Zephyr was rebooted after flashing the firmware before the NCS 2.8.0 port, but now, we have to manually reboot the boards. While not a critical issue, this reduces developer productivity, so I'd like Zephyr to be rebooted automatically after flashing the firmware.

@benedekkupper Please make this a priority.

benedekkupper, 2024-12-19:
Please try by adding this modification to `.vscode/settings.json`:
```json
    "nrf-connect.applicationOptions": {
        "${workspaceFolder}/device": {
            "flash": {
                "softreset": true
            }
        }
    },
```

mondalaci, 2024-12-19:
Works, thanks!

kareltucek, 2024-12-20:
It seems this fixes the issue for vscode only. 

How does one reboot automatically when compiling from command line?

benedekkupper, 2024-12-20:
Add the `--softreset` argument to the `west flash` command

kareltucek, 2024-12-20:
Alright, thanks!