UltimateHackingKeyboard/firmware #1051: Simplify BLE pairing

mondalaci, 2024-12-20:
Consider the current pairing popup on Android:

[image]

After typing the pairing code on the UHK, I must tap the Pair button.

On some other keyboards, this pairing popup doesn't contain a Pair button but only a Cancel button. When the pairing code is entered, the popup disappears, and the pairing succeeds. The UHK should follow suit.

mondalaci, 2025-01-25:
@benedekkupper Can you look into this? Users keep contacting us, and we keep telling them they must hit the Pair/Connect button on various OSes, which shouldn't be necessary if this issue was implemented.

mhantsch, 2025-01-25:
I can't even hit a pair button because none appears. I can't pair the UHK80 to my Linux laptop unless I resort to some commandline foo, which also only works sporadically.

benedekkupper, 2025-01-27:
Fixed via #1102

closed 2025-01-27 by benedekkupper