UltimateHackingKeyboard/firmware #1052: Make CI disallow warnings

mondalaci, 2024-12-20:
I want a stricter build process that disallows warnings, which should result in fewer bugs and cleaner build output. Please configure CI accordingly.

The current warnings:

```
/home/laci/projects/firmware/firmware/device/build/uhk-80-right/hid-usage-tables/src/hid-usage-tables/src/hid/parser.py:34: SyntaxWarning: invalid escape sequence '\s'
  page_regex = re.compile('(?P<id>[0-9a-fA-F]{4}) "(?P<name>.*)"\s*')
/home/laci/projects/firmware/firmware/device/build/uhk-80-right/hid-usage-tables/src/hid-usage-tables/src/hid/parser.py:124: SyntaxWarning: invalid escape sequence '\-'
  filename_regex = re.compile('(?P<id>[0-9a-fA-F]{4})\-(?P<name>[a-zA-Z0-9_\-]*)\.txt')
/home/laci/projects/firmware/firmware/device/build/uhk-80-right/hid-usage-tables/src/hid-usage-tables/src/hid/codegen/common.py:21: SyntaxWarning: invalid escape sequence '\s'
  id = re.sub('[\s]+', '_', id)
```

```
/home/laci/projects/firmware/firmware/device/src/usb/usb.cpp:157:40: warning: '%s' directive argument is null [-Wformat-overflow=]
  157 |         printk("USB config changing to %s\n", magic_enum::enum_name(conf).data());
      |                                        ^~
```

Would be nice to fix, but I'm not even sure if these are warnings:

```
Partition 'mcuboot' is not included in the dynamic resolving since it is statically defined.
Partition 'mcuboot_pad' is not included in the dynamic resolving since it is statically defined.
Partition 'mcuboot_primary' is not included in the dynamic resolving since it is statically defined.
Partition 'mcuboot_primary_app' is not included in the dynamic resolving since it is statically defined.
Partition 'settings_storage' is not included in the dynamic resolving since it is statically defined.
```

I'm unsure whether the following is a single warning, and I assume the noisy part will be fixed by the upcoming NCS 2.9.0 update:

```
Zephyr build without c2usb
CMake Warning at /home/laci/projects/firmware/zephyr/CMakeLists.txt:952 (message):
  No SOURCES given to Zephyr library: ..__firmware__boards__ugl__uhk-80

  Excluding target from build.

CMake Deprecation Warning at /home/laci/projects/firmware/nrf/cmake/partition_manager.cmake:32 (message):
  

            ---------------------------------------------------------------------
            --- WARNING: Child and parent image functionality is deprecated   ---
            --- and should be replaced with sysbuild. Child and parent image  ---
            --- support remains only to allow existing customer applications  ---
            --- to build and allow porting to sysbuild, it is no longer       ---
            --- receiving updates or new features and it will not be possible ---
            --- to build using child/parent image at all in nRF Connect SDK   ---
            --- version 2.9 onwards.                                          ---
            ---------------------------------------------------------------------
Call Stack (most recent call first):
  /home/laci/projects/firmware/zephyr/cmake/modules/kernel.cmake:255 (include)
  /home/laci/projects/firmware/zephyr/cmake/modules/zephyr_default.cmake:142 (include)
  /home/laci/projects/firmware/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/laci/projects/firmware/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:12 (find_package)
```

mondalaci, 2024-12-20:
Please let's not make CI disallow warnings, as Karel doesn't prefer that, but please let's fix as many warnings as possible with this PR.

kareltucek, 2024-12-20:
(It would be nice if CI distinguished errors from warnings and reported them accordingly. I am not aware of a reasonable way to achieve that though.)

closed 2025-08-07 by mondalaci