UltimateHackingKeyboard/firmware #1057: Touchpad troubles.

kareltucek, 2024-12-24:
https://forum.ultimatehackingkeyboard.com/t/latest-firmware-disabled-numerous-functions/1846/5

pcooke9, 2024-12-24:
Here's my configuration files. I'm using a fresh install of Agent Release v5.0.1 with the bundled firmware (v12.1.0). I was testing this with mostly default config settings.

I made changes to:
•key/display LED brightness lowered to 20
•keystroke delay raised to 10ms
•touchpad pinch-to-zoom divisor raised to 16
•touchpad hold continuation timeout raised to 1000

UHK60v2 config:
[UserConfiguration.json](https://github.com/user-attachments/files/18238596/UserConfiguration.json)

UHK80 config:
[UserConfiguration.json](https://github.com/user-attachments/files/18238599/UserConfiguration.json)

___

Something else I noticed; when the UHK80 has its halves joined, the info at the top of Agent's "Firmware" tab intermittently shows the message: **"Touchpad is detected, and its firmware is implemented by the right keyboard half."**

It shows in seemingly random intervals (between 4-30 seconds). Usually only appearing briefly, but sometimes the message remains for several seconds.

Obviously the touchpad is not connected when the UHK's halves are joined...

cumanzor, 2024-12-26:
[3966727881.json](https://github.com/user-attachments/files/18249457/3966727881.json)

My issue is specifically that when I do a click and drag, regardless of how long I'm holding the tap and drag, it gets "sticky" so that I have to tap the trackpad again for it to stop dragging.

kareltucek, 2024-12-26:
Thanks guys! Should be fixed by #1064

pcooke9, 2024-12-26:
> Thanks guys! Should be fixed by #1064

Firmware v12.1.0 #85a7083 indeed seems to have fixed both single-click and two-finger right-click issues on the UHK60 and UHK80.
___

Still experiencing the intermittent message: **"Touchpad is detected, and its firmware is implemented by the right keyboard half"** when no Touchpad is connected. Is that something to be worried about? Just seems a little curious.

kareltucek, 2024-12-27:
> Still experiencing the intermittent message: "Touchpad is detected, and its firmware is implemented by the right keyboard half" when no Touchpad is connected. Is that something to be worried about? Just seems a little curious

Ah, thanks for the headups. I don't think you need to be concerned, but I guess I should investigate it.

Sounds like the halves are trying to communicate with their modules and are talking to each other instead. Touchpad module is the only one affected probably because its protocol consists almost solely of data reads.

cumanzor, 2024-12-27:
Thanks! Do I need to build/compile the fw file myself for now?

closed 2024-12-27 by mondalaci

mondalaci, 2024-12-28:
No. You can flash the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12509029737 build from Agent or wait for the next firmware release.

cumanzor, 2024-12-28:
@mondalaci got it, we are back to normal, thank you so much!

PR #1064 (2024-12-27): Fix touchpad presses.
Closes #1057