UltimateHackingKeyboard/firmware #1058: All LEDs must be lit in switch testing mode

mondalaci, 2024-12-24:
```
erase-hardware-config.ts
factory-update-uhk60.ts firmwareDir uhk60v2
write-hardware-config.ts uhk60v2 ansi
set-variable.ts testSwitches 1
```

After executing the above but before pressing any key, the ANSI LShift LED should be enabled, but it's not on the UHK 60.

[image]

mondalaci, 2024-12-27:
Now all the LEDs light up, but then switch testing only works on the right half.

Pressing a key on the right half makes the LEDs of both halves go out, which is inconsistent with the behavior of the UHK 80 on which the halves work separately in this respect. Afterward pressing keys only on the right half make their LEDs light up, not on the left half.

To reproduce this, the hardware configuration must be erased according to the above instructions.

kareltucek, 2024-12-29:
Current state of affairs:

As far as I know, current implementation of the switch test mode in #1062 is correct.

The trouble goes *much* deeper - into KSD i2c stack. Right half doesn't receive the key count i2c message, but it *thinks* it received it, while the rx buffer remains unchanged. Since the buffer contains valid last message, its crc is valid, and hence it is parsed as a valid response for key count, which is how right half ends up believing that left half has exactly one key, and therefore reads only the first from subsequent keystate queries.

Invalidating the crc after message read doesn't solve the issue: it looks like the i2c driver actually writes the same old message into the rx buffer second time. Therefore it is also possible that left half actually sends the old (i.e., moduleId) message twice.

Ways forward:
1) I can hack-fix validation of this specific message with an ugly hack so that whenever it receives key count 1, it treats the message as invalid, and requests the message again.
1) We can try to figure out the actual cause of the issue. For this the next step is capturing an actual oscilloscope reading of the communication to at least identify whether it is the right or the left half that is the culprit, and then delving into the internals of the KSD code.
1) We can globally amend our uhk module slave protocol so that it checks whether the message is actually a response to the query that we think it is a response for. This would render modules unusable with current uhk80s, as a module firmware upgrade would be needed.
1) "Fix" the factory process by reenumerating or powercycling the right half after the last hardware config write in the factory procedure.

I am not at all a fan of 1. 

I don't have tools and experience to efficiently employ 2. Given our backlog of high priority-issues, this doesn't sound as a justified use of my time.

I am a fan of 3, but we can't do that now because of our unability to upgrade uhk80 modules.

Some combined approach could be used, such as applying 1 or 4 as a quick patch to unblock the factory, and employ 3 when we are able to upgrade modules.

@mondalaci your take?

---------------------

The evidence (more for my reference, feel free to skip; of course you are welcome to doublecheck):

Log of left half slave protocol i2c communication from the point of view of the right side's uhk module driver:

[image]

Testing with code at chance_test_led_ligth_all_behavior_logging branch 

Most notable change, left half module adds one control byte so that we see what is going on with the memory:
```
        case SlaveProperty_ModuleId: {
            TxMessage.data[0] = MODULE_ID;
            TxMessage.data[1] = 123;
            TxMessage.length = 2;
            break;
        }
        case SlaveProperty_KeyCount: {
            TxMessage.data[0] = MODULE_KEY_COUNT;
            TxMessage.data[1] = 234;
            TxMessage.length = 2;
            break;
        }
```

Testing with this script:
```
#!/bin/bash -x
# cat fw/package.json
./erase-hardware-config.ts --vid=14248 --pid=1 --usb-interface=4
echo "Please powercycle the board and press any key!"
read line
lp=`realpath .`
cd /opt/firmware/scripts
nvm use 20
./make-release.mjs --allowSha
cd $lp
./factory-update-uhk60.ts /opt/firmware/scripts/uhk-firmware-12.1.0 uhk60v1
```

And gathering logs via:
```
sleep 2; ./exec-macro-command.ts --vid=14248 --pid=1 --usb-interface=4 "printStatus"
```

kareltucek, 2024-12-29:
Actually, I have been digging into the crc... ...and turns out that crc of an empty message is zero, which obviously isn't ideal, as a zero message contains zero as its crc.

In the light of that a reasonable fix seems to be to require any message to simply be nonzero in addition to having a valid crc.

PR #1062 (2024-12-31): Light entire matrix at the beginning of switch test mode.
Closes #1058
closed 2024-12-31 by mondalaci