UltimateHackingKeyboard/firmware #1059: Mouse keys drifting/repeating

Zetaphor, 2024-12-25:
I upgraded to firmware 12.1.0 a couple of days ago and I've had a few incidents of the mouse keys driting/repeating.

What I mean by this is I will only tap a mouse key direction and the mouse will instead move a massive amount, continuing beyond the time that I've released the key. The first time I noticed this behavior I tried again by only tapping the key, and it moved a much larger distance than a tap should have emitted.

The first time this happened I disconnected both halves from each other and the USB cable, reset them both, and that resolved the issue. This time I tried resetting both haves, but leaving them both connected via the bridge cable and USB, this did not resolve the issue. Once I fully disconnected and reset everything the issue was resolved.

I was not having this issue before this firmware version, I've had it three times since upgrading.

I keep both halves connected via the bridge cable and the right half connected to USB.

kareltucek, 2024-12-25:
Please share your user config.

Zetaphor, 2024-12-25:
[UserConfiguration.json](https://github.com/user-attachments/files/18246490/UserConfiguration.json)

Zetaphor, 2025-01-02:
I'm not sure if this should be listed here or a new issue but I just had a pretty severe software breakdown.

Here is the series of events:

1) Flash to the new 12.2.0 firmware that was released yesterday
2) After the firmware upgrade I again had severe mouse key drifting
3) As I do whenever this happens I disconnect USB, disconnect the bridge cable, and reset both sides, starting with the right
4) The left side went solid white, and then started slowly flashing the keys solid white. Normally when I reset it just goes solid until I reconnect the bridge cable
5) Reconnect, still solid white. Did a few more resets of both halves, at one point in the process the left half was working, but would only respond to a few keypresses before it just stopped responding
6) Downgrade to 12.1.0 and quite a few more resets and now everything is working fine

pcooke9, 2025-01-03:
> * The left side went solid white, and then started slowly flashing the keys solid white. Normally when I reset it just goes solid until I reconnect the bridge cable
> * Reconnect, still solid white. Did a few more resets of both halves, at one point in the process the left half was working, but would only respond to a few keypresses before it just stopped responding

I haven't noticed any similar mouse key drift, BUT...

I just discovered that if I reset the UHK80's left half while the spiral cable is connected, it will cause it (left half) to go into a disconnect/reconnect loop (with the LEDs slowly flashing all white). I get the Windows device disconnect/reconnect sound and the UHK80's OLED display left battery indicator flashes, so I know it's actually disconnecting. If I disconnect/reconnect the spiral cable during the connection loop, it will stop. The right half stays normal when the left half is looping.

This can be reproduced every time I reset the left half with the bridge cable connected. After the loop is stopped, the left half stays stable (even when the bridge cable is connected), so I don't think it's a bad cable or other hardware failure.

If I reset the left half with the bridge cable disconnected (both halves connected wirelessly), it restarts with no problems.

UHK80 with firmware v12.2.0

kareltucek, 2025-01-03:
@pcooke9 this sounds like yet another problem. I don't see why left half reset should cause right half disconnect from the computer.

In any case, this seems like one of the firmwares is crashing :-(. We really need to implement the zephyr shell so that at least some logs can be pulled out, although I am not sure if it will be possible to retrieve the crash logs via usb :-[.

kareltucek, 2025-01-03:
Is it a new v12.2.0 regression, or a v12.1.0 regression?

pcooke9, 2025-01-03:
> this sounds like yet another problem. I don't see why left half reset should cause right half disconnect from the computer.

Sorry, I don't think the right half is disconnecting from the PC. I have both halves connected to the PC via USB, and the left half is disconnecting/reconnecting and causes Windows to make the device connection sounds from the left half USB. The UHK's OLED display only shows the left battery indicator flashing, not the right battery indicator.

> Is it a new v12.2.0 regression, or a v12.1.0 regression?

I was experiencing this issue with v12.2.0
I have a lot to do today, so I'll have to test with earlier firmware versions tomorrow.

kareltucek, 2025-01-03:
> I have both halves connected to the PC via USB

Ah, I didnt realize this!

pcooke9, 2025-01-06:
Ok, I tested the UHK80's left half connection loop issue with all the FW v12 release versions.
Each firmware was flashed with both halves connected to the PC via USB, and I was using the spiral bridge cable. I have some of the flash logs if you'd like me to post them.

**1. FW Release v12.0.1**
•After flashing, both halves were normal.
•The connection loop does not happen with this version.
___

**2. FW Release v12.1.0**
•After flashing, the left half LEDs were all white and non-responsive (right half was normal).
After a minute or so, the left half LEDs turned off for the letter keys, and the "Fn2" key and top row "ESC-F6" keys all changed to my normal base layer colors, but left half keypresses remained unresponsive.

•Disconnecting/reconnecting the bridge cable and pressing a couple keys on the left half caused the right half to restart itself, and both halves resumed normal operation.

•If I reset the left half with the button, the connection loop happens.
The left half LEDs stay solid white without flashing (FW v12.2.0 fades on/off white).
Windows device connection sounds happen, and the UHK80 OLED flashes the left battery indicator.

•Disconnecting the bridge cable seems to resume normal operation while the halves are connected wirelessly, but the connection loop with white LEDs returns if I reconnect the bridge cable.
•Disconnecting the bridge cable again, resetting the left half while both halves are connected wirelessly, and then reconnecting the bridge cable, resumes normal operation.
•After that, if I try resetting the left half again, it will go back into the connection loop.
___

**3. FW Release v12.2.0**
•After flashing, I had jumbled output with short bursts of repeating keys. (This doesn't happen every time I flash v12.2.0)
•Disconnecting the bridge cable and using the UHK80 with both halves connected wirelessly seemed normal, but jumbled output returned when reconnecting the bridge cable.

•The connection loop also happens with this version. The left half LEDs all turn white and fade on/off while Windows makes the device connection sounds, and the UHK80 OLED flashes the left battery indicator.
•Disconnecting the bridge cable seems to resume normal operation while the halves are connected wirelessly, but the connection loop with white LEDs returns if I reconnect the bridge cable.
•Disconnecting the bridge cable again, resetting the left half while both halves are connected wirelessly, and then reconnecting the bridge cable, resumes normal operation.
•After that, if I try resetting the left half again, it will go back into the connection loop.

I thought this might be interesting to note from one of the FW v12.2.0 logs (I have no left module BTW):
```
[DeviceService] Left module firmware version:  0.0.0
[DeviceService] Current left module firmware checksum:  
[DeviceService] New left module firmware checksum:  be9c3aaaf3c3bb42a37f65caf5bd21c2
```

The logs for v12.0.1 & v12.1.0 show "Left module firmware 12.*.*" and have both current & new checksums.

___
Anyway, it seems like the left half just isn't properly flashing all the time. Sometimes the logs will say the left half was skipped, and other times the left half will say it was flashed successfully, but begin the connection loop immediately after flashing is done (without using the reset button).

I'd also like to note that I haven't noticed any immediate issues when I flash these firmware without the bridge cable connected. It will still have the connection loop when resetting the left half though. More often than not, there are weird quirks when I flash with the bridge cable. My bridge cable seems physically fine, and it's stable with firmware before v12.1.0.

I don't know if any or all of these things are related, or if they need to go into separate issues? 
To be clear, the only effect that's consistently reproducible for me, is the connection loop after resetting the left half with the button on the back (v12.1.0 & v12.2.0).