UltimateHackingKeyboard/firmware #1060: UHK80 with Windows is in disconnected state after waking up from sleep

n0rbertg, 2024-12-25:
Hi,

Keyboard: UHK 80
FW version: 12.1.0
Connected via USB
OS: Windows11

When Windows wakes up from sleep the keyboard is in a disconnected state. After reconnecting the USB cable, it works again.

mondalaci, 2024-12-25:
https://forum.ultimatehackingkeyboard.com/t/unable-to-wake-up-uhk-80/1857

multjiang, 2024-12-26:
Yes, I was the one posted in the forum and have the same issue.

kareltucek, 2024-12-26:
Here are some additional details from testing (with Windows):

Zephyr shell:
```
# put it to sleep
uhk80:right$ L2 Suspend
 === set transport null
L0 ON
L state 0
L0 ON
L2 Suspend
L0 ON
L state 0
L0 ON
L2 Suspend

#wake it using power button
uhk80:right$ L0 ON
L state 0
L0 ON
L2 Suspend
L0 ON
```

Generally, the state reporting is pretty quirky. 

@benedekkupper would you look into it please? Maybe it makes sense to do together with https://github.com/UltimateHackingKeyboard/firmware/issues/1021 (I will follow up shortly on what is needed.).

benedekkupper, 2025-04-22:
Those who are affected please test this firmware (current master branch): https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14422644330/artifacts/2933383354

multjiang, 2025-04-23:
> Those who are affected please test this firmware (current master branch): https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14422644330/artifacts/2933383354

Seems fixed now 👍 I'll test again tomorrow

closed 2025-04-24 by mondalaci