UltimateHackingKeyboard/firmware #1065: Upgrading from 12.0.1 to 12.1.0: multiple issues

jwr, 2024-12-27:
I'm filing this as a separate issue, because I'm not 100% sure if this is the same problem, and because this time it's clear that a specific version change made things much worse.

Upgrading from 12.0.1 to 12.1.0 made my UHK80 unusable and I had to downgrade. The symptoms described in #1004 were amplified and happened almost all of the time. Additionally, and I'm not sure if it is the same problem or not, Cmd-TAB for switching between applications behaved strangely, as if cmd was held, or delayed, I am not quite sure. I wasn't able to get Cmd-SPACE at all.

I downgraded to 12.0.1 and these symptoms went back to previous behavior, e.g. multiplied key presses happen only sometimes.

Please feel free to close as duplicate if you think another issue better describes this.

mondalaci, 2024-12-28:
Do you use your UHK 80 via USB, BLE, or the dongle?

In any case, it'd be useful to unpair every BLE device, disconnect the dongle, and try to reproduce the issue via USB.

jwr, 2024-12-28:
This was via USB. I didn't get the bluetooth connection to work yet (filed another issue for that), and I have not tried the dongle.

jwr, 2024-12-28:
See https://github.com/UltimateHackingKeyboard/agent/issues/2475 for an example of erratic behavior — a message typed with 12.1.0.

mondalaci, 2024-12-28:
@kareltucek The reported behavior hasn't been reported by anyone so far. Any insights? Jan's sample text:

> i retire dht eupgrade and hti stime i towrke,d as you can see from this message, which i"M ytping as is after th eupgrade. After i selected 'force', obht halves were upgraded. nOw shift owrks erratically, as well as other cahracetrs as you can see ;_) i htink i might have ot odnwgrade...

kareltucek, 2024-12-28:
Nope :-(

mondalaci, 2024-12-28:
@kareltucek We should provide a way for users to wipe the settings partitions on the halves and the dongle. I'm suspicious that Jan's UHK 80 keeps trying to connect to a BLE device, resulting in the mentioned behavior.

#995 would also be very useful to debug such issues.

mhantsch, 2025-01-04:
I think I mentioned a few times that I believe it is useful - especially in problem cases - to be able to wipe the memorised bluetooth profiles / security keys. I had a few cases where the computer could not connect to the keyboard anymore because for whatever reason they had disagreeing keys.

mondalaci, 2025-01-04:
@mhantsch Agreed. Just reopened #1029.

closed 2025-09-16 by kareltucek