UltimateHackingKeyboard/firmware #1066: Unable to use Bluetooth

jwr, 2024-12-27:
Trying to use Bluetooth with a Mac.

Open Bluetooth Settings, click on "Connect" button next to unpaired "UHK80". Enter pairing code (why? I would prefer only to confirm the code).

UHK80 moves to "My Devices", shows "Connected" and a battery status of 100%.

Disconnect USB.

The display on the UHK still shows "USB Cable". No keypresses reach the computer.

Disconnect coiled cable.

The display goes black, then when a key is pressed the keyboard re-awakens. The display now shows "disconnected". No keypresses reach the computer.

Reconnect coiled cable.

The display still shows "Disconnected", no keypresses reach the computer.

Firmware 12.0.1.

mhantsch, 2025-01-04:
Entering the pairing code is OK, but to get my UHK80 to enter pairing I have to manually run these commands:

```
(base) max@max-framework:~/src/ergobat$ bluetoothctl devices
...
Device D2:89:0A:CA:61:05 UHK 80
...
(base) max@max-framework:~/src/ergobat$ bluetoothctl connect D2:89:0A:CA:61:05
Attempting to connect to D2:89:0A:CA:61:05
AdvertisementMonitor path registered
(base) max@max-framework:~/src/ergobat$ bluetoothctl trust D2:89:0A:CA:61:05
[CHG] Device D2:89:0A:CA:61:05 Trusted: yes
Changing D2:89:0A:CA:61:05 trust succeeded
AdvertisementMonitor path registered
(base) max@max-framework:~/src/ergobat$ bluetoothctl pair D2:89:0A:CA:61:05
Attempting to pair with D2:89:0A:CA:61:05
AdvertisementMonitor path registered
```

Only after entering this sequence of commands, I can then switch the UI page on Pop_OS for the UHK80 to "connected", and then it will pop up the pairing code, and the UHK80 will finally prompt me to enter the pairing code. After this, the keyboard works over bluetooth.

But when I then connect the UHK80 via USB, and open the "Connections" page, the BLE connection is not listed. Shouldn't it show up there?

[image]

mhantsch, 2025-01-04:
If I don't go through that manual process, I can't get the UHK to connect/pair to my Laptop. I see it in the Bluetooth panel, it's listed as "Not set up". When I click on it, a spinner appears for about 1 second, and then it goes back to "Not set up".

mondalaci, 2025-01-04:
> But when I then connect the UHK80 via USB, and open the "Connections" page, the BLE connection is not listed. Shouldn't it show up there?

Not yet, but we're working on it:
- https://github.com/UltimateHackingKeyboard/firmware/pull/1031
- https://github.com/UltimateHackingKeyboard/agent/pull/2450

jwr, 2025-01-05:
> As for the core issue, Karel is reworking the relevant part of the firmware, and I'll give more feedback to him soon.

Ok, so for now Bluetooth is expected not to work? I wasn't sure, that's why I reported this, so that we have an issue open.

mondalaci, 2025-01-05:
@jwr After the

> UHK80 moves to "My Devices", shows "Connected" and a battery status of 100%.

step you mentioned, I'd assume that you can use your UHK 80 via BLE. Can't you? If not, please elaborate.

The UHK 80 already works via BLE, but there are related missing features and bugs.

jwr, 2025-01-05:
Perhaps I wasn't clear in my report — at no point after disconnecting USB did any keypress reach my computer. In other words, I am unable to use BLE.

mondalaci, 2025-01-05:
Gotcha! So, the BLE connection is established between your Mac and UHK 80, but your UHK doesn't switch to it.

We'll provide a way to debug this issue, and more importantly, we'll allow the deletion of BLE data on the keyboard side, hopefully resolving this issue. Also, we'll make the connection switchover more robust, fixing the core of the issue.