UltimateHackingKeyboard/firmware #1069: Mouse doesn't work on Android

mondalaci, 2024-12-28:
When pairing my UHK 80 to my Android phone, the scancodes are transmitted as expected, but not mouse HID events. After unpairing my UHK, mouse doesn't work via USB unlike before, only after rebooting the right firmware.

@kareltucek Can you reproduce it?

mondalaci, 2025-01-05:
@kareltucek Is this issue specific to c2usb?

mondalaci, 2025-01-30:
I haven't been able to reproduce this issue since 7ef5cccac528c7ccde05a2bcfaa92425d2026b4c was merged.

mondalaci, 2025-01-31:
I can sporadically reproduce this issue. The mouse interface didn't work over BLE a couple of minutes earlier, then I rebooted the right half with the reset button, and after pairing my UHK 80 with my phone, it worked again.

0xNineNineSix, 2025-05-12:
I have exact same issue on multiple laptops running windows 11.

Brand new UHK80:
- Right hand firmware 14.0.0
- Left hand firmware 14.0.0
- Key cluster firmware 14.0.0
- Trackpoint firmware: 10.4.0

Works perfectly fine wired. When I connect to either laptop via bluetooth, all the buttons work, but the trackpoint stops working, and the mouse layer commands stop working as well.

mondalaci, 2025-05-12:
@benedekkupper How can I provide you with useful debugging regarding this issue?

iprok, 2026-01-16:
> it would probably help him if you captured a wireshark log of the communication.

I would be happy to do it but I don't know how to capture bluetooth data using wireshark. (Pretty familiar with capturing network data) Could you please give me sort of short manual?

kareltucek, 2026-01-17:
I am afraid I don't know from top of my head. I am actually not sure if it is possible due to encryption and security concerns. Will look into it at some point.

I am now revisiting the stack and have noticed a couple of details that may make a difference, so hang on.