UltimateHackingKeyboard/firmware #1073: Side scrolling doesn't work on the UHK 80

mondalaci, 2025-01-04:
Left and right scroll doesn't work on the UHK 80, neither with mouse keys, nor with the modules.

@kareltucek Is this a c2usb needing Benedek?

Salivala, 2025-01-09:
Might end up being related but on some menus scroll down through trackpoint is spotty and inconsistent.

kareltucek, 2025-01-09:
- when writing uhk80 code, Benedek assumed that c2usb and uhk60 mouse report formats are binary equivalent, which was true at the time 
- this was broken at some point. I believe by high-res scrolling implementation for uhk60
- current state of affairs that uhk60 code uses 2 bytes per axis, while uhk80 only 1 byte, which of course the horizontal scroll events be written into vertical high byte scrolling. Therefore also those reports about strange scrolling behavior.
- Benedek's high-res PR bumps this to 2 bytes again

What to do about it: while there are multiple ways to fix it from my side, I think best to test and merge Benedek's high-res scrolling PR.

mondalaci, 2025-01-09:
> Might end up being related but on some menus scroll down through trackpoint is spotty and inconsistent.

@Salivala What do you mean by "on some menus"? Please elaborate so that we can reproduce it.

@kareltucek c2usb and KSDK should implement the same reports, whichever version is better. I don't know what makes one report format better than the other, and I'm interested in the details.

Salivala, 2025-01-09:
> @Salivala What do you mean by "on some menus"? Please elaborate so that we can reproduce it.
>
> @kareltucek c2usb and KSDK should implement the same reports, whichever version is better. I don't know [...]

Windows 11 menu -> System ( or any scrollable windows 11 display ) scroll down. Stuttery jerking when attempting to scroll down. Happens on both trackpoint scroll down and mousemode scroll down.

kareltucek, 2025-01-10:
You want 2 byte range for high res scrolling, that much is obvious.

pcooke9, 2025-01-10:
> > Might end up being related but on some menus scroll down through trackpoint is spotty and inconsistent.
> 
> @Salivala What do you mean by "on some menus"? Please elaborate so that we can reproduce it.
> 

I forgot all about this, but it happens to me as well. I was going to make an issue for it, but I thought it might be a duplicate of:
https://github.com/UltimateHackingKeyboard/firmware/issues/1039

I mentioned it with an included video [here](https://github.com/UltimateHackingKeyboard/firmware/issues/998#issuecomment-2543580878):
> In some apps on Windows 11, scrolling with the UHK80 is extremely erratic using **"uhk-firmware-11.2.0-stuck_keys_hunt_v4.tar.gz"**. It doesn't happen with **"uhk-firmware-11.2.0-stuck_keys_hunt_v2.tar.gz"**.
> 
> It occurs when using mouse keys scrolling and with the touchpad.
> 
> It seems fine on the UHK60.
> 
> Some apps I've noticed with strange scrolling behavior are: Windows Settings app, Microsoft Store, and Windows Start Menu > All Apps.
> 
> Scrolling works fine in Windows Explorer, internet browsers, etc..
> 
> **EDIT: This also happens with all master branch firmware versions I've tried since:** https://github.com/UltimateHackingKeyboard/firmware-uhk80/actions/runs/12169751543
> 
> **Here's a video showing the odd behavior:** https://www.youtube.com/watch?v=Eo7TZqElkDg
https://github.com/UltimateHackingKeyboard/firmware/issues/998#issuecomment-2543580878

As you can see in the video, scrolling is fine in internet browsers and Windows Explorer, but some Windows apps are wonky. 

Even more strange, I noticed that in the Windows 11 "Start Menu > All Programs" list, if you scroll down a bit with your non-UHK mouse and then try to scroll down some more with UHK80's mouse keys, it will only scroll in the upward direction; regardless of using either the "scroll down", or "scroll up" mouse keys. In the Windows Settings app, mouse keys won't scroll down at all. Weird.

___

Note again that IIRC, the last firmware that didn't have this issue was:
https://github.com/UltimateHackingKeyboard/firmware80/actions/runs/12169751543

pcooke9, 2025-01-10:
Both the side scrolling and other strange scrolling behavior in certain Windows apps seems to be fixed with:
https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12706975201

My touchpad scrolling was super slow, so I had to drop my Touchpad scroll speed divisor down from 8 to 1.

There still seems to be the occasional jerk in speed with this FW, but it's definitely at least mostly fixed. Maybe further tweaking of the touchpad settings will smooth things out. Also maybe this will help: https://github.com/UltimateHackingKeyboard/agent/issues/2487

kareltucek, 2025-01-16:
Should be solved #1047

closed 2025-01-16 by kareltucek