UltimateHackingKeyboard/firmware #1074: How to build the firmware

Makurisan, 2025-01-06:
Can you please give some instructions where you normally build the firmware.
I tried building it under WSL2 and Windows 11. On both envionment without results.

There is nothing menationed about this in the readme.

Thanks

kareltucek, 2025-01-06:
I use Ubuntu. Afaik, Laszlo and Benedek are on linux as well.

What exactly is wrong about the paths? Forward slashes, `:` in file names, or case?

Anyways these specific files are not important, so you can ignore the errors. Also feel free to fix them and open a PR.

Makurisan, 2025-01-06:
Not anyways. Give instruction which works and not that the user must do some things because it is not possible to checkout the source on the system.
Next thing is the "west" thing. Give instruction how to install.

kareltucek, 2025-01-06:
Sure! Are you after uhk60 or uhk80 specifically?

Makurisan, 2025-01-06:
I ordered the uhk60 v2.

kareltucek, 2025-01-06:
Ah, that simplifies matters greatly since you don't need zephyr at all. Unfortunately, the uhk60 build has issues on Windows too as far as I know, because paths, symlinks, etc..

I.e., working instructions should be:

- Install Ubuntu on a ext4 partition

- Follow following README.md sections:

    - Fetching the codebase

    - Old Minimal development setup

As for west, I am installing it from pip3: `pip3 install west`.

Makurisan, 2025-01-11:
The "./build.sh setup" finishes now successful but if I built 
"./build.sh uhk-dongle build" for example I get the following
 output and it is not clear if this is a success because of the error after zephyr.elf:
[image]

Then another question:
Is it possible to build the uhk60v2 firmware without "Kinetis Design Studio" and if yes, how?

mondalaci, 2025-01-11:
We're sorry, but the firmware stack can be difficult and time-consuming to set up, and we can't allocate resources to helping individuals.

closed 2025-01-11 by mondalaci