UltimateHackingKeyboard/firmware #1075: Left keyboard half issues.

Salivala, 2025-01-07:
Left half of keyboard appears to need the connecting cable even when connected. 
When disconnecting connector cable and reconnecting, left half seems to turn off and on repeatedly, need to unplug and reconnect right usb cable. 

Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

mhantsch, 2025-01-07:
Does your UHK80 have batteries?

pcooke9, 2025-01-07:
This sounds like the left half connection issues mentioned throughout this thread, starting from this comment here:
https://github.com/UltimateHackingKeyboard/firmware/issues/1059#issuecomment-2568171080

Salivala, 2025-01-07:
> Does your UHK80 have batteries?

no

Salivala, 2025-01-07:
They keyboard reconnect issue doesn't seem to be happening now. Not sure why.

Salivala, 2025-01-07:
another thing of note, for the cable needing to be there, the pins seem to be connecting to eachother fine; agent knows when they're together and they're apart, when the cable is attached, so it's using that cross data with the pins. On the UHK 60 you didn't need the cable when both halves are connected, but it's almost like there is no power transfer happening if it's just relying on the keyboard pins.

Salivala, 2025-01-07:
I also have modules for both sides and they both work properly.

pcooke9, 2025-01-07:
> False alarm, it's still happening.

Try disconnecting the bridge cable, reset the left half with the reset button on the back of the left half, and then reconnect the bridge cable after it finishes the reset.

That works to temporarily stop it for me.

If that doesn't help, try re-flashing the firmware without the bridge cable (both halves joined in wireless mode).

Salivala, 2025-01-07:
So this is expected behavior? I just assumed because the uhk60 could do this that the 80 would be able to. Doesn't really matter to me personally since i just keep the thing attached anyways but I'm just going through the board and seeing if everything is g2g before i get back to using it

Salivala, 2025-01-07:
in both instances my halves were together.

Salivala, 2025-01-07:
[image]
could it be due to the connectors being too far apart? with the keyboard together if i put a little pressure with my wrists to separate the halves it will also read as split.

mondalaci, 2025-01-07:
Unlike the UHK 60, the UHK 80 halves don't communicate and power each other via the pogo pins.

Please create a [new Agent issue](https://github.com/UltimateHackingKeyboard/agent/issues/new) for phantom modules and delete the original issue template when creating new issues.

mondalaci, 2025-01-08:
Closing as a duplicate of #1079.

closed 2025-01-08 by mondalaci