UltimateHackingKeyboard/firmware #1077: UHK80 sometimes lagging and sending duplicate keypresses (or wrong order)

n0rbertg, 2025-01-07:
Hi,

Strange issue and it's difficult to troubleshoot as I can't reproduce it reliably, but it happens quite often.

Setup:
* UHK80 with fw 12.0.1
* OS: Windows 11

I have two computers and use a USB switch to switch the keyboard between them. (I haven't tried the built-in feature yet, will do this eventually :))

Sometimes, when I change computers, it takes quite a while for the keyboard to wake up. It's not necessarily tied to the switch, but it happens more often after it. 

The keyboard becomes very laggy for about a minute or two, and it sends various signals to the host:
- When I press a single key, it just repeats it (like a single press sends 6-8 characters)
- The order of some key presses is swapped; it sends the one I pressed later first
- Sends the characters with a noticeable lag

As mentioned, this happens for a minute or two and after a while it settles down and gets back to normal. 

If someone could explain how to debug this behaviour, I am happy catch it and attach logs when it happens. I haven't seen anything like this with my UHK60 (v1)

kareltucek, 2025-01-07:
Please upgrade and retest with at least 12.1.0. We have upgraded to a newer SDK and Benedek has done a huge amount of work on USB since then.

kareltucek, 2025-01-07:
Alright, then that is of a concern.

Unfortunately in this case it is hard to rule out the switch as the source if the issue - we generally recommend using UHK without any intermediate devices in case of troubles :-(.

kareltucek, 2025-01-07:
(As for the built in feature, we are still working on it...)

n0rbertg, 2025-01-07:
Is there a way to debug this when it's happening? 

(As mentioned, I was using my UHK60 with the same switch without any issue)

kareltucek, 2025-01-07:
None accessible.

You can try capturing wireshark logs, and while it is possible that you will be able to spot something suspicious, it is unlikely for a mere mortal to be able to seriously debug them.

You can also try to pull some data from uhk directly, but we don't have much support for that yet.

If you have a jlink probe then you should be able to acces zephyrs shell, but that is not enough as most debug-relevant features are disabled at compile time for performance reasons, and enabling them usually requires writing custom code.

Exposing zephyr shell over usb is high on my priority list, but debugging usb over usb is likely to run into a wide variety of problems.

pcooke9, 2025-01-07:
This sounds a lot like what I was trying to explain here: https://github.com/UltimateHackingKeyboard/firmware/issues/998#issuecomment-2569517323

And here:
> **3. FW Release v12.2.0**
> •After flashing, I had jumbled output with short bursts of repeating keys. (This doesn't happen every time I flash v12.2.0)
> •Disconnecting the bridge cable and using the UHK80 with both halves connected wirelessly seemed normal, but jumbled output returned when reconnecting the bridge cable.
> 

I don't use a switch or hub; both USB halves directly into the PC.

I also suspect the repeating mouse keys reported by @zetaphor might be the same.

kareltucek, 2025-01-07:
Maybe try to see whether there is anything abnormal about the way uhk registers in the system, but again, I am not fluent with usb to help much 

@pcooke9 yes, sorry about that, I am quite overwhelmed these days.

pcooke9, 2025-01-07:
@kareltucek No worries, I can see you've got a lot going on. 🫣 It probably doesn't help that my comments are so long winded either. 🤣

kareltucek, 2025-01-07:
No worries 😉.

jwr, 2025-01-08:
FWIW< hti sis likely th esam eproblme that I reported in another issue. i mypin hi on hte latest firmware 912.2.00 and thi si swhat comes out. i iwll be downgradin gagain, i guess.

mY uhk I socnnecte div auSB, wih hte coiled iwre between th ehalves.

Also, each half resets itsel feveyr once in  ahwile, boin gdark an docming up again afer several seconds. THe resets seme to happen only if i"M ytping osmehting, leaving ht ekeyboard yb itself doens' tseem to tirgger resets.

mhantsch, 2025-01-08:
Same here. Wanted to use my UHK80 with firmware 12.2.0 today and got only garbled output, missing keypresses, and repeated characters, as well as the keyboard (at least right half) rebooting every 30-60 seconds. Completely unusable. Went back to 12.0.1 and it works again.

lightningRalf, 2025-01-09:
So i am still on 12.1.0 na dhav the sam eissues. fo r aaaaaaaafew days. i tis hpapnein gmost often avvvffter i ne dth eeeepc ot lseep and hten wak ei tup iwht  akeystroke.
Whne it waks up. ht eleft hlaf gets activated first. an dhtne comes hte seocnd. After that sometimes i towkrs iright, other itmes i tjumbles up htese htings.

as you can read in this message.
After some itme i tttdisconnects agian na dhtan reconnects, bu thte errors are still htere, but  alittle bit less.

After some time again, the keyboard disconnects again, and then reconnects and then all the issues are gone.
Just like now. (Started typing this message as soon as I COULD enter my password correctly and got to type it out) So it was about a timespan of around 5 minutes I imagine. 2.5 min for each 'restart' of the keyboard.

I connect the UHK80 directly to the PC.

mondalaci, 2025-01-09:
There are too many variables, so going forward, please every one of you let us know whether the keyboard halves are connected via the bridge cable, and whether you're using your UHK via USB, BLE or the dongle.

Zetaphor, 2025-01-09:
Every time I've encountered this issue I have had the bridge cable connected, and USB connected to the right half

n0rbertg, 2025-01-09:
Same, mine is connected via USB with the right half and using bridge cable

jwr, 2025-01-09:
Bridge cable connected, USB connected to the right half.

pcooke9, 2025-01-10:
Bridge cable and both halves connected with USB.

mhantsch, 2025-01-10:
Bridge cable and USB connection on the right half.

lightningRalf, 2025-01-10:
Bridge cable and USB connection on the right half.

(after updating to 12.2.0) I did not stumble upon that issue anymore.

tunney, 2025-01-10:
Bridge cable, usb to right side. 12.2 on a Mac

I occasionally see the gargles input. As if the left side lagged the right.

blach, 2025-01-13:
I also have missing key presses, but only if not using the bridge cable:

UHK 80, USB-C on right half, no bridge cable, 12.3.0, connected to Mac. The issue was also when using firmware 12.2.0.
It also happened with BLE instead of USB-C.

I don't seem to have missing key presses when connecting the two halves using the bridge cable.

vpal, 2025-01-13:
One strange thing. I received my UHK80 with the 12.2.0 firmware (this is what it came with from the factory).
I was using it for a week without seeing these issues, I did not flash the firmware during this time.
Now I upgraded to 12.3.0 and I started seeing similar issues after the keyboard wake up, right after the FW upgrade. 
It was just typing the letter 'e' endlessly and things like that. 
Right half connected via USB and two halves connected via the bridge cable.

I will keep testing and report my findings, will let you know if the issue persists.

n0rbertg, 2025-01-13:
One more symptom, when the keyboard acts messed up, the left shift doesn't work properly but the right one does. (left just randomly changes characters or not at all, acting erratic) - using bridge cable

mhantsch, 2025-01-14:
Further info: after upgrading my **UHK 80** to firmware 12.3.0 (from 12.0.1), the symptoms initially reappeared. 

Lagged keystrokes, keys getting stuck and autorepeating, keys swapped between left and right halves, layers (including Shift!) only activating 1-2 seconds after the keydown event (so subsequent characters would not be shifted, e.g. holding left shift and then tapping `/?` repeatedly would output `/////???????????`) and weird side effects with my hrm macros. Basically, the UHK 80 was unusable.

Then the keyboard rebooted itself (at least the right half) a few times, and suddenly all the problems went away and everything functions normally now. I didn't change anything in between, I just kept trying to use the keyboard and figure out how I create a reproducible sequence of keystrokes that shows the issues.

Not sure if that helps at all. Just trying to describe what I experienced. Now I am using 12.3.0 and it seems to work now.

mhantsch, 2025-01-14:
The symptoms came back after saving a configuration :-(

I have one more observation: when the symptoms appear, as soon as I disconnect the bridge cable, and the UHK switches over to wireless connection, the lag disappear, and for example the left shift and `/?` issue goes away.

When I reconnect the bridge cable, the issues reappear.

Or let me say, with the bridge cable disconnected, the issues are much beter. There is still the occasional misisn letter or swap, as you can see in this vey sentence.

With hte cable ocnnecetd, the tyipng is completely garblele ld ma

mhantsch, 2025-01-14:
I rebooted the lleft half a few times to see if the problem goes awya but it keeps reappearing.

is laso reboote dhte rihg thalf but htat doesn't hel peihte.r

pcooke9, 2025-01-14:
I can reliably reproduce the garbled output by simply resetting the right half with the button on the back while the bridge cable is connected. It doesn't happen if you reset it without the bridge cable. It doesn't matter if I use only one or both halves connected to the PC via USB. This is reproducible 100% of the time for me on master branch firmware v12.3.0 #c9fa41a. I can check with earlier versions if you'd like.

**Steps to reproduce and temporarily resolve:**
1. Make sure bridge cable is connected, and reset the UHK80's right half using the reset button on the back.
2. Observe garble output, repeating keys, etc.
3. Disconnect bridge cable and observe normal output while left/right halves are communicating wirelessly.
4. Reconnect bridge cable and observe garbled output, etc.
5. Disconnect bridge cable and reset the UHK80's right half with the reset button on the back while left/right halves are communicating wirelessly.
6. Observe normal output while bridge cable is still disconnected.
7. Reconnect bridge cable and observe normal output as well.
8. If you reset the right half again while the bridge cable is connected, it will scramble the output again...

tunney, 2025-01-15:
For whatever reason otday htis i shappening ot me now and laos random disocnnects from hte computer.
COnnected iva birdge cable

This kjeyboard i sno t useable nad was released afr too soon.

kareltucek, 2025-01-15:
@pcooke9 I believe that what is going on is that when the reset button is pressed, the board reloads into bootloader, and stays there for 5 or so seconds. During this time left half indeed doesn't work. 

But once left half boots (i.e., when leds light up), everything starts working fine for me. Do your issues persist after the leds light up?

pcooke9, 2025-01-15:
> But once left half boots (i.e., when leds light up), everything starts working fine for me. Do your issues persist after the leds light up?

Beginning with the UHK80 in a properly working state;

1. I reset right half with the bridge cable connected, and wait for the full reboot cycle (LEDs and OLED return).

2. Then garbled output is induced, and _continues_ until I disconnect the bridge cable and reset right half again.
After that full reboot cycle, normal operation is restored (no garbled output, with or without bridge cable).

pcooke9, 2025-01-15:
I'm curious when most people first notice the scrambled output. It seems a lot of the reports state it happens after upgrading firmware. If so, do they flash with the bridge cable connected, or disconnected?

Most of my issues tend to happen after flashing firmware with the bridge cable connected. I very rarely have troubles if I flash without it. I don't think I've seen the garbled output happen unless something causes the UHK80 to reset first. That includes when it has trouble waking from sleep, after flashing firmware, or whenever it spontaneously resets on its own.

To be clear, my UHK80 is mostly stable. I just happened to figure out a way to intentionally induce the garbled output (on my UHK, at least) in hopes it might be helpful.

Over the next day or so, I think I'll focus on testing the consistency of anomalies when flashing firmware.

pcooke9, 2025-01-15:
> But once left half boots (i.e., when leds light up), everything starts working fine for me. Do your issues persist after the leds light up?

Quick question...
When resetting the _**RIGHT**_ half, should the _**LEFT**_ half also reset (with its LEDs turning off/back on)? Because my UHK80 doesn't do that. The left half LEDs stay continuously lit. I don't remember if it was different a few firmware versions ago or not.

mhantsch, 2025-01-15:
As far as I know, the halves reset separately. Rebooting the right half does NOT automatically restart the left half.

pcooke9, 2025-01-15:
Ok, scared me for a minute.😅

kareltucek, 2025-01-15:
I am wondering if elmag interference could be the culprit - messages over uart are not (yet) checked for consistency.

mondalaci, 2025-01-15:
Oh no, we really should use CRC and resend invalid packages, as the bridge cable is not shielded! :)

If you folks only experience this issue with the spiral cable but not without it, then we probably have the culprit.

pcooke9, 2025-01-15:
Is there a easy way to test interference? If a super cheap EMF meter would suffice, I wouldn't mind grabbing one. I'm sure it'd be fun to play with anyway.

I usually keep my phone at least a couple feet away from my board, so I doubt that's it for me.

mondalaci, 2025-01-15:
I put my phone as close to the bridge cable as possible while having a call, and I couldn't reproduce this issue. I don't have a better idea, and I expect EMI to vary wildly across environments/devices.

mhantsch, 2025-01-15:
> If you folks only experience this issue with the spiral cable but not without it, then we probably have the culprit.

I experience LOTS of issues when the cable is connected. Without it, I experience the OCCASIONAL missing letter. There may be multiple issues involved, or delays introduced over the serial link cause a small race condition to trigger substantially more often.

mondalaci, 2025-01-15:
By "serial link", I assume you mean the bridge cable. The wireless connection between the keyboard halves is also a (wireless) serial link. Just wanted to clarify to avoid miscommunication.

mhantsch, 2025-01-15:
Yes, the bridge cable.

tunney, 2025-01-15:
> I am wondering if elmag interference could be the culprit - messages over uart are not (yet) checked for consistency.

I use my UHK80 in the same environment and situations as my UHK60s and have done for the last 3-4 years. I have never experienced elmag interference in this environment with the UHK60.

I have experiences elmag interference in other environments with the UHK60 but never in the home office. So the UHK80 would have to be significantly more sensitive than the UHK60.

mondalaci, 2025-01-15:
> So the UHK80 would have to be significantly more sensitive than the UHK60.

The UART of the UHK 80 should be a lot less prone to interference than the I2C bus of the UHK 60. CRC backs the I2C on the firmware level, unlike UART. We have yet to see how UART performs with CRC.

tunney, 2025-01-15:
I cannot replicate elmag interference with phone usage near the UHK80, I could with the UHK60

mhantsch, 2025-01-15:
> the bridge cable is not shielded

I really don't think it's elmag interference, or not only. We could try wrapping the bridge cable in aluminium foil and grounding it, just to see if it makes any difference.

vpal, 2025-01-15:
One thing that came in to my mind is if this is related to either battery or the modules.
I'm on 12.3.0 and I have only seen a similar issue once when I upgraded to 12.3.0, not after that.
There is a good test that was shown on the forum in a video with the LEDs changing when pressing one of the layer activators. I also tested it that way and for me it seems to be full in sync.
Connected via USB cable and the bridge cable also being connected.
Is there anyone who has this without battery or any module connected?

Hope this helps the investigation.

n0rbertg, 2025-01-15:
> Is there anyone who has this without battery or any module connected?

I don't have battery nor any module and I still experience this.

pcooke9, 2025-01-16:
> > the bridge cable is not shielded
> 
> I really don't think it's elmag interference, or not only. We could try wrapping the bridge cable in aluminium foil and grounding it, just to see if it makes any difference.
> 

Are shielded RJ9 connectors even a thing? I don't think I've ever seen one.

kareltucek, 2025-01-16:
Guys, could you please test [uhk-firmware-12.3.0-uart-fixes.tar.gz](https://github.com/user-attachments/files/18441047/uhk-firmware-12.3.0-uart-fixes.tar.gz)?

(In case of invalid messages it may output an error - in that case the oled shows a warning triangle. You can retrieve the message from Agent. The content isn't important, but the presence and number of these messages is of interest.)

tunney, 2025-01-16:
as in the count of the lines here?

[image]

kareltucek, 2025-01-16:
Yes! Or more precisely how often they appear, and whether the problems persist, whether you notice any issue the moment before it appears 😊

tunney, 2025-01-16:
> Yes! Or more precisely how often they appear, and whether the problems persist, whether you notice any issue the moment before it appears 😊

How does one clear the error messages so that one can tell when the message is a new one?

kareltucek, 2025-01-16:
The clearStatus macro command

tunney, 2025-01-16:
Sorry, silly question I even have that one already assigned....

tunney, 2025-01-16:
I am seeing loads of lags and freezing from the left side but one one "Invalid UART message received!" message

Bridge cable, and USB connection.

kareltucek, 2025-01-16:
Hmm, I will add some more logging...

tunney, 2025-01-16:
Really bad freezing this even. Really frequent and persistent. No more events though.

tunney, 2025-01-17:
No freezing or lagging in the last forty minutes but one error:

[image]

tunney, 2025-01-17:
Wow i continue to be amazed at how bad the UHK80  can be. Just when I think that it cannot get worse......

Left side is basically unusable this afternoon. Lack of responsiveness is astounding.

Another "invalid UART message received and the left side takes tens of seconds to respond.  The display on the keyboard  cycled between the wifi icon, connected icon, and an X 

Staying on a UHK60v2 for the foreseeable.

kareltucek, 2025-01-17:
Can you please try [uhk-firmware-12.3.0-uart-fixes-v2.tar.gz](https://github.com/user-attachments/files/18458780/uhk-firmware-12.3.0-uart-fixes-v2.tar.gz)?

This time I explicitly log uart timeouts, this time on the oled. 

I assume you will see lots of uart timeouts.

[158 more replies]

closed 2025-03-06 by kareltucek