UltimateHackingKeyboard/firmware #1078: UHK80 stays disconnected during boot

vpal, 2025-01-07:
During boot my UHK80 stays disconnected.
This means during UEFI/BIOS stage and also when GRUB loads.
The keyboard is only connected when the operating system boots.

This is a significant issue as it is not possible to:
- Log into UEFI settings.
- Select different boot options.
- Upgrade firmware.
- Type disk encryption password.
- Etc.

This is a Dell Latitude 3520, so pretty standard, hopefully quality hardware.

Other keyboards (for example QMK firmware) work during the boot phase.

Please let me know if you need more details.

vpal, 2025-03-02:
Greetings, any news on this? It is getting frustrating over time that:

- Keyboard does not wake up the computer when on the computer is on sleep.
- Most of the times when the computer is hibernated and it is woken up from hibernation (not via the keyboard as that does not work) the keyboard needs to be reconnected to be used.
- Not able to use keyboard during boot, not able to do anything in BIOS (EFI) and not able to do anything in GRUB, not even when the keyboard is reconnected.

My gut feeling is that these are related to the same issue, but if you would like me to open an issue for any of the mentioned problems let me know.

kareltucek, 2025-03-02:
@benedekkupper, according to your knowledge, are these c2usb issues or uhk issues?

kareltucek, 2025-03-02:
@vpal, feel free to give a shot to https://github.com/UltimateHackingKeyboard/firmware/actions/runs/13520273276/artifacts/2648024398 (from https://github.com/UltimateHackingKeyboard/firmware/issues/1143#issuecomment-2681754914 ) It should make ihk work fine i  bios.

Aso for the other two: can you reproduce them of you suspend/hibernate theachine and then try to wake it up immediately (i.e., within seconds or minutes)?

vpal, 2025-03-02:
@kareltucek

- Wake up: It does not work with any of the firmwares (12.3.5, 12.3.5 #42ab22b) , I tried right after putting the laptop to sleep and then pressing some keys, but no reaction, the keyboard states that it is connected after putting the laptop to sleep.
- Boot time keyboard connection: The same with the firmware in the link as before. The keyboard states that it is disconnected until the OS boots up. So no entering BIOS and no typing encryption password in GRUB unfortunately.

vpal, 2025-03-06:
Okay, so I did some testing. The sleep wake up issue is probably not related to this one.
I remembered that it worked when I got the keyboard and now I downgraded to 12.2.0 and with this version the wake up works as expected. When I put the laptop to sleep and press the keyboard it wakes up. I tried right after putting it to sleep and also after an hour or so and both work fine, so this worked in the past but then some regression came (probably right after 12.2.0) and then it stopped working.
For the boot time keyboard availability and the hibernation issue I haven't seen any firmware it is working with.
I will open a new issue for the wake up regression.
For the disconnected state it seems there are already around for issues open.

kareltucek, 2025-03-07:
> are you actually able to reproduce this issue on your side

Yes.

When I just reboot the laptop, uhk80 still works fine throughout boot. But if I powercycle it, then it doesn't start working until gdm login screen.

vpal, 2025-03-08:
> > are you actually able to reproduce this issue on your side
> 
> Yes.
> 
> When I just reboot the laptop, uhk80 still works fine throughout boot. But if I powercycle it, then it doesn't start working until gdm login screen.

Thanks a lot. Hopefully that helps with a quick fix :)

mondalaci, 2025-03-12:
When I reboot my i9-14900K / Z790 Linux PC, I can enter its UEFI BIOS with my UHK 80 running firmware 12.4.0. Even mouse navigation works, but mouse buttons don't. When I reboot my UHK 80 in the BIOS, it keeps working.

Since I can't reproduce the issue, I'll rely on you guys to test upcoming fixes.

vpal, 2025-03-15:
@mondalaci, thanks for checking.
🤞🤞🤞 that it doesn't take too long, it is pretty inconvenient.

benedekkupper, 2025-04-22:
Those who are affected please test this firmware (current master branch): https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14422644330/artifacts/2933383354

vpal, 2025-04-23:
> Those who are affected please test this firmware (current master branch): https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14422644330/artifacts/2933383354

I did a few tests and it is much better now. Even if I shut down my laptop fully and start it again I'm able to log into BIOS and type my password.
I also seem to be able to confirm that hibernating my laptop and waking it up does not leave the keyboard disconnected, though I would like to do some more tests here.

What does not work for sure: If I'm in BIOS or in the GRUB prompt and I press Ctrl + Alt + Del after the restart the keyboard does not work anymore, but then if I shut down the laptop and start it again it works again.

This is already very big progress and fixes most of the pain points.

vpal, 2025-04-23:
One other thing that I tested and it does not work is waking up a suspended computer with the keyboard, but maybe this firmware is not supposed to fix that.

vpal, 2025-04-24:
Okay, did some additional tests.
The above statements (except for the hibernation related ones) are related to my Dell laptop.

HP laptop:

- I have not found a way to get into the BIOS, it works from the laptop keyboard
  - When the laptop is shut down fully and started again it seems like the keyboard is still booting up when the machine expects the input that triggers entering BIOS (but this is just an assumption, see additional points). It seems this input is needed right after the laptop is powered on.
- When the laptop is hibernated and I start it up
  - When I don't touch the keyboard, the keyboard is available when the OS starts up
  - When I try to press around buttons during boot time I have to reconnect the keyboard
- When the laptop is fully shut down and I start it up
  - When I don't touch the keyboard, the keyboard is available when the OS starts up
  - When I try to press around buttons during boot time I have to reconnect the keyboard

When I press around buttons on the keyboard during boot time (and a reconnect is needed) I see a `Disconnected` message for some time during the boot phase. When I don't press anything I don't see this message and I do not need to reconnect the keyboard.