UltimateHackingKeyboard/firmware #1083: down scrolling via hotkeys / mod key / UHK 80

lightningRalf, 2025-01-09:
similar to the #1039 
I have the following issue: 
When trying to use up and down scrolling with microsoft whiteboard and excalidraw.
The up scrolling works as expected.
But when trying to scroll down. It always scrolls to the left and down in a 45 degree angle.

mondalaci, 2025-01-09:
Can you reproduce with mouse keys and/or modules? Do you use your UHK via USB, BLE, or the dongle?

pcooke9, 2025-01-10:
I just checked Microsoft Whiteboard with Touchpad and mouse keys using:
https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12706975201

It works fine now.😁

lightningRalf, 2025-01-10:
> Can you reproduce with mouse keys and/or modules? Do you use your UHK via USB, BLE, or the dongle?

I can reproduce this via the scroll module (left side). I am using the UHK with USB(on the right keyboard half).
I only have the trackpoint. Do not know, how to enable scrolling with the trackpoint yet unfortunatly.

lightningRalf, 2025-01-16:
> I just checked Microsoft Whiteboard with Touchpad and mouse keys using: https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12706975201
> 
> It works fine now.😁

sorry for the naive question, but I couldnt find anywhere how to update the firmware by itself.

Here is what I tried:
1. Download the latest release.
2. DeZIP it.
3. In the agent: choose firmware file and flash it
4. looking for the right file in the following folder: boards/ugl/uhk-80 / couldnt find it there.
4b. also looked for the *.gz and *.bz2 extensions via search. but didnt find anything either (on windows with windows search)

Any recommendations for me?

lightningRalf, 2025-01-16:
> I just checked Microsoft Whiteboard with Touchpad and mouse keys using: https://github.com/UltimateHackingKeyboard/firmware/actions/runs/12706975201
> 
> It works fine now.😁

I just downloaded the 12.3.0 and installed it.

Problem is still existing, with both Microsoft Whiteboard + Excalidraw (both online versions) (I am on Windows)

pcooke9, 2025-01-16:
> I just downloaded the 12.3.0 and installed it.

The fix hasn't been implemented in the official release version yet. It's part of the "high-res-scrolling" branch:
https://github.com/UltimateHackingKeyboard/firmware/actions?query=branch%3Ahigh-res-scrolling

I think there's still a bit of work to be done on other aspects of the high-res scrolling (speed issues, etc.), but the last I checked, the issue you posted here was fixed. If I'm not mistaken, I think there needs to be an update to Agent as well as the firmware before high-res scrolling is implemented in a future release.

I could be wrong, but it looks like it's not too far away...   https://github.com/UltimateHackingKeyboard/firmware/pull/1047

mondalaci, 2025-01-16:
Please clarify the "beeping sound" because I don't know where it originates. Perhaps a video would be the most useful.

lightningRalf, 2025-01-16:
https://github.com/user-attachments/assets/aa928346-7d88-4fcf-89e0-984130a6fda0

mondalaci, 2025-01-16:
Thank you, but I'm still in the dark. I don't know whether this sound is operating system-specific or application-specific. Reproducing it with a mouse or any other pointer device would be useful.

lightningRalf, 2025-01-16:
> Thank you, but I'm still in the dark. I don't know whether this sound is operating system-specific or application-specific. Reproducing it with a mouse or any other pointer device would be useful.

I also have the key cluster module. On the scroll ball it works flawlessly. - Also with the mouse there is no issue to be found.

Thanks for getting back to me. 
The recording is from Obsidian Excalidraw plugin.
When I am trying to reproduce it in the Excalidraw.com: each key individually do work flawlessly, when pressing two keys at the same time, the same error occur.
When trying Microsoft Whiteboard (online version): the individual keys cause problems.

Hope this helps?
I am open to try different things, if you could suggest things to test.

mondalaci, 2025-01-17:
I'm trying to get to the bottom of this, but statements like "when pressing two keys at the same time, the same error occur", or "the individual keys cause problems" are way too vague.

I understand you experience issues with the "Obsidian Excalidraw plugin", also called Excalidraw.com, and Microsoft Whiteboard (online version). We need to know the exact issues, and please try to reproduce the issues with other keyboards and mice.

lightningRalf, 2025-01-17:
Thanks for your feedback. I am trying to be more percise.

### Environment
OS: Windows 11
### Browsers:
Microsoft Edge
Brave
### Recording Software: Windows Snipping Tool
### Hardware Setup
UHK Keyboard with Key Cluster Module
### Testing Controls:
Multiple physical mice: Working correctly
Key Cluster scroll ball: Working correctly
### Core Issue
Interference pattern occurs when combining:
Mouse emulation key
Directional inputs (scroll up/scroll down/ scroll left/ scroll right)

> I'm trying to get to the bottom of this, but statements like "when pressing two keys at the same time, the same error occur", or "the individual keys cause problems" are way too vague.
With the 'pressing two keys' I meant, pressing: mouse emulation key + Scroll up AND scroll left at the same time.

### Critical Observation
Screen recording appears to exacerbate the issue:

Windows Snipping Tool triggers interference pattern more rapidly
Manifests as:
Screen jitter (see video)
System beep sound (see video)
[image]

https://github.com/user-attachments/assets/11541b55-a59d-4a7e-934c-c9c7d972d806

I hope this clarifies the problem?

pcooke9, 2025-01-17:
I'm on Windows 11, and I don't experience any system beeps using the Microsoft Store app version of Whiteboard (or any beeps at all, that I've noticed).

lightningRalf, 2025-01-17:
> I'm on Windows 11, and I don't experience any system beeps using the Microsoft Store app version of Whiteboard (or any beeps at all, that I've noticed).

do you use the mouse modifier key + the scroll keys? 

For me, it extends to any scrolling on the web browsers and also outlook. (but only when using the keys as described in the post above)

mondalaci, 2025-01-17:
Thanks! I'm on firmware 12.3.1 and can reproduce this issue with the UHK 60 and 80.

Vertical scrolling gets very slow when using mouse keys simultaneously in horizontal and vertical directions.

Horizontal scrolling also gets very slow when scrolling northwest and southeast but remains normal when scrolling northeast or southwest.

I could reproduce the above issues on Windows but not Linux or Mac.

However, when disabling scroll axis locking and scrolling in a circle with my trackpoint module on every OS, scrolling feels jittery, so I think modules are also affected, and this issue affects every OS.

mondalaci, 2025-01-17:
A useful testing page for this issue:
https://home.tobias-schwarz.com/tests/viewport/scroll-both.php

pcooke9, 2025-01-17:
> do you use the mouse modifier key + the scroll keys?

I've never heard that sound on my PC at all.
Do you know for sure what app is making the sound? Is it the browser/app, or the actual Windows System sounds? If you're not sure, could you check the Windows Volume Mixer and mute any apps on-by-one to find out?

I tried to replicate it with the online version of Whiteboard and Excalidraw, but it still doesn't happen. I tried with the UHK80 mouse keys and the touchpad module (my only module), as well as my non-UHK mouse.

[image]

___

@mondalaci Do you hear the beeping in Windows as well?

pcooke9, 2025-01-17:
> I'm on firmware 12.3.1 and can reproduce this issue with the UHK 60 and 80.
>
> Vertical scrolling gets very slow when using mouse keys simultaneously in horizontal and vertical directions.
>
> Horizontal [...]

I was about to report the same, but you beat me to it.

mondalaci, 2025-01-17:
Thanks for confirming! I don't hear the sound, but I only tested in browsers.

lightningRalf, 2025-01-18:
> > do you use the mouse modifier key + the scroll keys?
> 
> I've never heard that sound on my PC at all. Do you know for sure what app is making the sound? Is it the browser/app, or the actual Windows System sounds? If you're not sure, could you check the Windows Volume Mixer and mute any apps one-by-one to find out?

Thanks for the hint with switching the sound of.
I tested in on multiple apps. (claude desktop, discord and microsoft edge) all volume turned of, beeping sound is still there, on all three) So it is the system.
And not only when scrolling in multiple directions, but also only up and down. for example on this github issue. But only starting, after I am on the both keys for about 0.5 seconds.

mondalaci, 2025-01-18:
The beeps may signify that Windows is flooded with excessive scroll events. Do [these commands](https://www.reddit.com/r/techsupport/comments/3btb7x/mousewheel_scrolling_causing_beeping_sound/) disable beeping?

```
sc stop beep
sc config beep start= disabled
```

lightningRalf, 2025-01-18:
> The beeps may signify that Windows is flooded with excessive scroll events. Do [these commands](https://www.reddit.com/r/techsupport/comments/3btb7x/mousewheel_scrolling_causing_beeping_sound/) disable beeping?
> 
> ```
> sc stop beep
> sc config beep start= disabled
> ```

Oh wow, thank you
This also solves the scrolling issues when scrolling up and down.

So now I only have the multidirectonal scrolling issue, that persists. 💖

pcooke9, 2025-01-19:
Wow, that's super strange. I don't believe I've ever seen that before. 🤷‍♂️ Hopefully there's nothing more serious going on with your system.

kareltucek, 2025-09-16:
Is this ticket still relevant? If so, let me know and I will reopen it...

closed 2025-09-16 by kareltucek