UltimateHackingKeyboard/firmware #1085: UHK80 and BT connection issues

tunney, 2025-01-10:
The BT connectivity on the UHK 80 has issues reconnecting to a Mac Book Pro M3 MAX

I connect the keyboard over BT, it pairs and is usable over BT. I then disconnect it and have it connected wired. I can still use it. However, the keyboard cannot then be connected and is not visible.

[image]

Flashing the firmware will allow me to reconnect to the keyboard over BT once I repair

Steps to reproduce:
1) Connect wired
2) Add UHK80 via BT and pair
2) Disconnect UHK80
3) Attempt to reconnect the UHK80
4) Keyboard will not appear in the list of available devices, is greyed out and does not have the option to connect present

[image]

blach, 2025-01-13:
I had this problem with firmware 12.2.0, but it seems to be fixed with 12.3.0:

Previously, the keyboard would not reconnect to the Mac when it was shutdown and rebooted. With 12.3.0 I just tried it again and it did reconnect, so I could even login with FileVault without connecting to USB-C. 🎉

tunney, 2025-01-13:
Now there is a whole new issue.
Connect it over BT, try to disconnect the device, watch it immediately reconnect.........

mondalaci, 2025-01-13:
@tunney But is it a new issue, or an [already existing one](https://github.com/UltimateHackingKeyboard/firmware/issues/990)?

tunney, 2025-01-13:
Fair enough.
Wasn't exposed in 12.2.0, probably masked by the other BT bug. F

mondalaci, 2025-01-13:
I'm closing this issue, then, assuming it's resolved.

closed 2025-01-13 by mondalaci