UltimateHackingKeyboard/firmware #1087: UHK 60v2 Requires Re-plugging After Mac Wakes from Sleep

Recordedfieber, 2025-01-12:
### Description
When my MacBook goes to sleep and wakes up, the UHK 60v2 keyboard partially wakes (lock screen appears), but the backlighting and LEDs stay off, and the keyboard becomes unresponsive. I need to unplug and re-plug the USB cable for it to work again.

### Steps to Reproduce

Connect UHK 60v2 to a MacBook (directly or via a hub).
Allow the Mac to sleep or turn off the display.
Press a key to wake the Mac.
Observe the keyboard behavior: no backlighting, LEDs, or responsiveness.

### Environment

UHK Model: UHK 60v2
Firmware: 
 Right keyboard half firmware: UltimateHackingKeyboard/firmware: 12.2.0
 Left keyboard half firmware: UltimateHackingKeyboard/firmware: 12.2.0 
OS: macOS
Connection: Direct or via USB hub

### Additional Notes

Similar issue reported when connected to a monitor's USB hub.
Possibly started after firmware update although the problem was persistent throughout the previous fw versions (12.0.0)

**Impact**

Not critical but inconvenient. Affects workflow by requiring re-plugging.

jsec, 2025-01-12:
Same issue, environment details below:

UHK Model: UHK 60v2
Firmware:
Right keyboard half firmware: UltimateHackingKeyboard/firmware: 12.2.0
Left keyboard half firmware: UltimateHackingKeyboard/firmware: 12.2.0
Key Cluster: UltimateHackingKeyboard/firmware: 12.1.0 (firmware upgrade for the cluster failed when updating to 12.2.0, unrelated to this ticket)
OS: macOS
Connection: USB-C through Apple Studio Display

vpal, 2025-01-14:
I see something similar with the UHK80.
When I put my computer to sleep the keyboard still says connected.
When I press a key the computer does not wake up.
I tested this on a Dell laptop with Linux and an HP laptop with Windows, the behavior is the same.
I started on 12.2.0 (the FW my UHK80 came with) and I have not seen this, it behaved like it supposed to behave.
After I upgraded to 12.3.0 I started seeing this behavior where it seems the keyboard is not aware anymore that the computer it is connected to went to sleep.
Should I open a separate issue for this?

The keyboard is connected via USB and the two halves are connected via the spiral cable.

kyoshiro, 2025-03-19:
Same here. Connection lost even when not locking the screen, after timeout of round about 3 minutes.

**Environment**

UHK Model: UHK 60v2
Firmware:
Right keyboard half firmware: UltimateHackingKeyboard/firmware: 12.4.0
Left keyboard half firmware: UltimateHackingKeyboard/firmware: 12.4.0
OS: macOS 15.3.2 / Apple MBA M1
Connection: Direct USB-C/Thunderbold

Using the same machine with Asahi Linux does not have this behavior. Maybe it's more related problem between USB-C/Thunderbold in MacOS than UHK itself.

madone550, 2025-04-25:
Just received my UHK 60v2 today and I am seeing this issue as well

 Right keyboard half firmware: UltimateHackingKeyboard/firmware: 13.0.2
 Left keyboard half firmware: UltimateHackingKeyboard/firmware: 13.0.2
 Key cluster firmware: UltimateHackingKeyboard/firmware: 13.0.2
 Touchpad is detected, and its firmware is implemented by the right keyboard half.
Mac OS 15.3.2 (24D81)
Apple MBP M2 Max

It happened when I had the keyboard plugged into my Apple Studio Displays rear USB-C.  I have moved it to OWC TB Pro Dock USB port now to see if it repeats

mondalaci, 2025-04-25:
My UHK 60 v2 is running firmware 13.0.2 and is connected to my MacBook Air M1 (2020) directly via USB. I close the lid, and the UHK lights go out. After 5 minutes, I open the lid, and the lights come back on; my UHK is now usable with the MacBook without needing to reconnect it.

Please help me reproduce the issue.

kareltucek, 2025-09-18:
Does the same patch just without the PowerMode_ActivateMode line resolve the issue? 

(PowerMode_ActivateMode should be actually called from a usb callback once host wakes up and signals UHK that it should resume. If waking host will succeed but the usb doesn't get woken up, it means that we should focus on usb stack. If it doesn't, then probably there is an issue with the power management logic.)

ds300, 2025-09-18:
yes it seems to still work!

kareltucek, 2025-09-18:
Please gather logs with this firmware. Go through a sleep wake cycle, then start Agent. A yellow pane should pop up. Copy the content here.

[uhk-firmware-15.1.0-wake_hunt_v1.5.tar.gz](https://github.com/user-attachments/files/22411591/uhk-firmware-15.1.0-wake_hunt_v1.5.tar.gz)

ds300, 2025-09-18:
I’m away from my keyboard until October now. Will try this and get back to
you then.

kareltucek, 2025-09-22:
Alright. So to sum up: 

- WakeHost is always called properly.
- Mac always wakes.
- But depending on either the WakeHost call site or WakeHost timing or something else things may go wrong, which probably means that one of the following happens:
  - mac doesn't send the resume usb signal
  - or mac sends it, but uhk fails to receive it
  - or it receives it, but some race condition happens which makes it to not update its power state

Would you mind a few more testing rounds to try to get to the bottom of this?

If so, please give a try to this build: 

[uhk-firmware-15.1.0-wake_hunt_f_v2.0.tar.gz](https://github.com/user-attachments/files/22467650/uhk-firmware-15.1.0-wake_hunt_f_v2.0.tar.gz)

Expected behavior: 
- After suspending your mac and tapping the key, the mac should wake up, but uhk not.
- At this point please start Agent with another keyboard. 
- Now either a trace will pop up, or it won't. Let me know which it is, and in the former case the trace.