UltimateHackingKeyboard/firmware #1088: The unit will not turn on after failing to update the Firmware.

bucchi-kitamura, 2025-01-14:
Description
I attempted to upgrade the firmware from 12.1.0 to 12.2.0 via the UHK Agent.

However, when I performed the upgrade, I encountered an error as shown in the attached image. Now, the keyboard won't power on, and I'm at a loss.

Could you please advise me on what steps to take next?

[image]

Environment
UHK Model: UHK 60v2

Agent version: 5.0.2
Firmware:
Right keyboard half firmware: UltimateHackingKeyboard/firmware: 12.1.0
Left keyboard half firmware: UltimateHackingKeyboard/firmware: 12.1.0
OS: macOS
Connection: Direct or via USB hub

Additional Notes

Impact

mondalaci, 2025-01-14:
Please see https://ultimatehackingkeyboard.com/knowledgebase?q=unbrick

bucchi-kitamura, 2025-01-16:
Thank you!

I will try to initialize the keyboard using this place as a reference.
https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half

I don't have the tools right now, so it will likely be over the weekend.

kareltucek, 2025-09-16:
Closing due to lack of followups :-).

closed 2025-09-16 by kareltucek