UltimateHackingKeyboard/firmware #1090: UHK60 v2 getting stuck on infinite key repeat

ajh3, 2025-01-14:
Sometime in the last couple weeks, my UHK60 v2 started to get stuck several times a day, infinitely repeating a keypress. When this happens, the last keystroke repeats forever until I press another key on the keyboard. This happens with single key strokes as well as chorded ones e.g., ctrl-v. There is no clear pattern to what causes it or when. Seems random.

This is not key chattering. It is strictly a case of getting stuck on infinite repeat.

Unfortunately I can’t confidently say which firmware version this started with. I do check and install firmware updates every few days, and can recall 2-3 firmware updates in the last month. So it started sometime in there. Although I am by no means certain it's a firmware problem, it might be a hardware problem with my UHK.

In the meantime, I downgraded to firmware 12.0, and if the problem goes away, I will start slowly moving back up.

Per the request in https://forum.ultimatehackingkeyboard.com/t/uhk-v2-recently-getting-stuck-on-infinite-key-repeat/1935, I have attached a copy of my user config.
[UserConfiguration.json](https://github.com/user-attachments/files/18412478/UserConfiguration.json)

kareltucek, 2025-01-15:
Hm. Pretty vanilla config. This is probably due to the huge refactor that was done in order to optimize the firmware for uhk80 and battery consumption. 

In any case, 11.2.0 should be stable. I suggest downgrading there. This issue will get solved eventually, but at this point we need to focus on uhk80. (My guess is that uhk80 is affected too, just masked by more severe problems.)

mhantsch, 2025-01-15:
Firmware 11.2.0 is rock solid. I have kept it (and the corresponding Agent 4.2.1 - make sure you disable Agent auto-update) on my UHK60s. I'll upgrade them when firmware 12.x becomes more stable.

ajh3, 2025-01-15:
Ok thanks, I'll downgrade to 11.2.0 and let you know how it goes.

As a data point: I went down to 12.0 yesterday and didn't have any problems all day, but then this morning it started happening again.

mondalaci, 2025-01-15:
There's no way, but if you have an older configuration in Agent (on the Configuration page) that you saved with 11.*.* or earlier, you can restore it.

ajh3, 2025-01-15:
I do, but my config has changed significantly since then. If it's going to be a while before this gets fixed, then I will probably try to recreate my current config in the old agent/firmware.

adandersen, 2025-01-15:
i received my UHK 80 in the mail two days ago, and the first time i used it it     wass doing  thatttttttt. It wwwwill alos reverse keys      on me as well. But it only sticckss for a short amount of timeeeeeeeeeeeeeeeeeeeeeee. It started up again just barely. If i stop typing for awhile it seems to fix itself. This part i didn't have to keep correcting like the first sentence as its working again. I stopped typing for maybe 1 minute and the sticky keys stopped while typing this out. Its only the left side of the keyboard that has the problem, the right side works fine as far as i can tell when it happens. I swapped my space to the left side as well, so that's why its also repeating spaces.

ajh3, 2025-01-15:
@adandersen it's not clear to me that your issue is the same. For me the keys repeat forever, they don't stop. And I haven't had any reversed input order.

adandersen, 2025-01-15:
i've had them repeat a couple times forever, but mostly its what i described

kareltucek, 2025-01-15:
> Is there any easy way to edit the file I posted to make it importable?

The formats are very similar. Shouldn't be too hard to compare the json structures and delete whatever is surplus there. Mostly deleting stuff. Of course given that you use a good editor that can at least work with folds well.

You can also use source code as a reference of what got changed - grepping right/src for VERSION_AT_LEAST should you give all the interesting sites.

ajh3, 2025-01-15:
Thanks, I was able to get my current config set up on Agent 4.2.1 / Firmware 11.2.0. Turned off auto updates for agent. 

Fingers crossed the problem stops, and it's not a hardware issue with my UHK. (My original UHK60 v1 failed due to a somewhat different key chattering issue, strongly hoping my v2 isn't dying as well.)

ajh3, 2025-01-17:
So far so good on 11.2.0!

jbstinson, 2025-01-20:
Got a UHK80 and I just set it aside and went back to my UHK60v2. My 80 doesn't have a battery and I always use direct connection. How bad is gonna mess things up if I go back to 11.2.0 on the 80?

kareltucek, 2025-01-20:
You mean on the 60? (11.x doesn't support uhk80, you cant flash it onto uhk80, agent wont allow it)

If you have a 11.2 userconfig backup, then the downgrade can be slightly painful, but otherwise it should be fine.

jbstinson, 2025-01-20:
@kareltucek thank you. I know it was a stupid question, but I really want to use the 80. When I am presenting at work and the 80 stops working and I have to cycle the 80 is just frustrating. I will just use the 80 for personal and not work for a bit. :) Its a work in progress and I will treat it as such.

dubiousgit, 2025-03-01:
I'm also on the UHK60v2 and getting this issue, as well as characters sometimes not pressing. I thought I was perhaps going senile and my typing was getting atrocious, but it's definitely just not registering some characters (and other times sending the same character forever, until I hit a different key.) I'm going to downgrade to see if it helps.

dubiousgit, 2025-03-01:
Downgraded to 11.2.0 and using Agent 4.2.1 for config reasons, will report back on how it goes.

ajh3, 2025-03-01:
> Downgraded to 11.2.0 and using Agent 4.2.1 for config reasons, will report back on how it goes.

I can tell you that I'm still on this combination, and with it the problem has not happened even once.