UltimateHackingKeyboard/firmware #1093: UHK 80 - ctrl + w does not work

lightningRalf, 2025-01-16:
Today after starting the PC new, I have a problem (again) with ctrl + W.
All other hotkeys, including ctrl + whatever work.

Also as you can see, the letter w also does work.
The combination of ctrl + w just doesnt close any windows.

I just flashed the firmware to 12.3.0. The issue still persists.

mondalaci, 2025-01-16:
Do you use your UHK 80 via USB, and are the halves connected with the bridge cable?

mhantsch, 2025-01-16:
Are you using the left or the right control key? Does it make a difference which key you are using?

lightningRalf, 2025-01-16:
I am using UHK 80 via USB and the halves are connected with the bridge cable.

But I think it is unreleated to the UHK and the firmware. And there are other issues with my setup.
Will close it for now, until I investigate further.

closed 2025-01-16 by lightningRalf