UltimateHackingKeyboard/firmware #1096: No Dongle connection after Pairing

opie4624, 2025-01-16:
I've paired the dongle via the Agent. Now how do I get the keyboard to connect using the dongle? After pairing, when I disconnect the USB cable, the display just shows "Disconnected" while the light on the dongle is blue. I've tried with both the 12.2.0 firmware and 12.3.0 from the github repo. Neither seems to make the dongle connect with the keyboard.

I see the dongle listed in the Connections section of the Agent.

While on the 12.3.0 firmware, I tried creating two smartmacro's with `switchHost next` and `switchHost previous` assigned to keys on the fn layer. These don't actually seem to do anything as I have two different devices (including the one with the Dongle) paired via BLE, but after disconnecting there seems to be no way to make the keyboard reconnect.

Deleting the dongle from connections and pairing it again (several failures before success) allowed it to connect.

[image]

[UserConfiguration.json](https://github.com/user-attachments/files/18444032/UserConfiguration.json)

opie4624, 2025-01-16:
It doesn't. It remained in a disconnected state, with the dongle light blue, even after pushing the reset button on both keyboard halves.

pcooke9, 2025-01-16:
This happened to me when upgrading to FW v12.3.1.

I deleted the dongle in Agent's Connections tab, reset the UHK80's right half with the bridge cable disconnected (just in case), and then Agent gave me the yellow banner at the top with the option to re-pair the dongle. The macros to switch devices worked fine after that.

kareltucek, 2025-09-16:
Sorry! 

I assume by this time everything is fine? If no, please let us know!