UltimateHackingKeyboard/firmware #1098: UHK 80 - 12.3.0 - keyboard disconnected after sleep

lightningRalf, 2025-01-16:
So the problem is the following: 
After having updated to 12.3.0. UHK connected via USB (right side) no battery, halves connected via cable. Windows

When sending the PC to sleep.

I can press any key (during the PC sleeps) the keyboard light turns on and the OLED shows 'Querty for PC - My PC'.
BUT nothing else happens! (expected behavior: PC wakes up)

When waking the PC up with the mouse, the keyboard light appears but the OLED shows: 'Querty for PC - Disconnected.'
After reconnecting the USB cable. It works normally.

Updating to 12.3.1 did'nt change any of the behavior.

mondalaci, 2025-01-16:
This is probably a duplicate of #1087.

lightningRalf, 2025-01-17:
> This is probably a duplicate of [#1087](https://github.com/UltimateHackingKeyboard/firmware/issues/1087).

Nope, difference being:
- My issues started with 12.3.0 - 12.2.0 worked.
- backlighting, LED & OLED work. But I cannot awake the PC.

vpal, 2025-01-18:
> This is probably a duplicate of [#1087](https://github.com/UltimateHackingKeyboard/firmware/issues/1087).

Yes, exactly same issue here with UHK 80.
@mondalaci, just one thing regarding the duplicate.
In the other issue the reporter says that the issue exists with 12.2.0.
For me 12.2.0 worked on the UHK 80, but a started seeing this with 12.3.0.