UltimateHackingKeyboard/firmware #1100: [BLE] Random disconnects / flickers

ndekross, 2025-01-18:
Keyboard disconnects after some time of use. In bluetooth device settings, keyboard flickers on/off (see video below) and the LED display shows it as disconnected. Windows 11 and firmware 12.3.1. The only remedy is to remove the device and reconnect.

https://github.com/user-attachments/assets/a340fe65-d21c-4e90-bf90-e585399bd3ff

mondalaci, 2025-01-18:
Does this happen after a specific even, such as waking up your PC?

#1056 may be related.

ndekross, 2025-01-19:
> Does this happen after a specific even, such as waking up your PC?

I think, last time it happened after fully charging and removing the USB-C cables.

ndekross, 2025-01-20:
This actually happened again just now, after a wake up.

mondalaci, 2025-01-21:
@kareltucek Do you think it's a c2usb issue?

michaeldesu, 2025-01-30:
I've noticed the same issue after powering on my laptop - I'm using firmware 12.3.2. I can't remove the device either via Windows bluetooth manager.. I suspect I need to reset the keyboard..

mondalaci, 2025-01-31:
I've managed to reproduce the issue in https://github.com/UltimateHackingKeyboard/firmware/issues/1056#issuecomment-2628422095

We'll make #1056 a priority, but it's currently not the highest priority issue.

earthling-slavlius, 2025-10-18:
Firmware 14.2.1-15.2.0 I had the same issue with the Intel cards AX210 and AX211 on my dell latitude 5340. I tried different bluetooth drivers. I had to go back to the version 22.200.0.2 from 15/12/**2022** and it is working now without this problem.