UltimateHackingKeyboard/firmware #1104: Issue with Bluetooth Searching

were, 2025-01-28:
I am having a UHK80 with firmware 12.3.1.
This keyboard is having issue with Bluetooth Searching with my macOS 14.6.1.
Every time, after disconnection, my MacBook cannot find and connect it from the existing device.
I need to forget the device and reconnect.

mondalaci, 2025-01-31:
I'm running firmware 12.3.2 and macOS 14.7.2. Please use these versions.

If I explicitly disconnect the UHK 80 on my Mac, it immediately reconnects according to #990, a known bug.

If I bring my MacBook far from my UHK 80 and then return, it disconnects and reconnects without any issue.

Does this issue persist with the above versions? If so, please elaborate on how to reproduce it.

mondalaci, 2025-02-10:
Closing issue due to the lack of feedback.

closed 2025-02-10 by mondalaci