UltimateHackingKeyboard/firmware #1105: UHK60 v2: Right side inoperable for all 12.x.x firmware versions

hcwesson, 2025-01-28:
Attempting to upgrade the firmware for my UHK60 v2 using the latest Agent release (`5.0.2` at the time of occurrence) to any of the `12.x.x` series results in the right half (and connected trackball module) failing to register input/key presses on my system, and all key LEDs on that side going dark. The left half works as usual.

When flashing any of the affected firmware versions, Agent indicates all modules were flashed successfully (left, right, trackball, key cluster). Logs report no errors.

I initially suspected that the right half was bricked and was preparing to follow the [official right-half unbricking procedure](https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half), but by downgrading Agent to `4.2.1` and flashing firmware version `11.2.0` I was able to restore to the previous working state (i.e. it doesn't seem like the right half was actually bricked—just incompatible with the `12.x.x` firmware).

So far, I'm able to consistently reproduce the problem with all v12 firmware versions using Agent `5.0.2`.

kareltucek, 2025-01-28:
Does the problem persist when you reset the user configuration? (Of course be sure to make a backup.)

kareltucek, 2025-03-07:
Closing for lack of replies.

closed 2025-03-07 by kareltucek