UltimateHackingKeyboard/firmware #1106: Fn "Stuck" when changing Keymaps

shiroikumakoori, 2025-01-29:
So i have a default Keymap , DVO.
I'll switch between many other keymap , including a QWE one.
by Fn + 1  or 2  or 3  , the Fn use is the default Fn position.

in order to change back to DVO ill do a Fn + 1 . 

the Fn in question is the " clickly" button below the default Mod button ( the left space bar ).

before latest frimware update i had always been doing this with zero issue. 
aftre this update oddly when it successfuly swap back to my DVO keymap. it will auto be stuck in Fn mode. i have to press Mod or Fn to cancel it out .

if i attempt this same sequence with othre button that i map as Fn , including the right "clicky" button it wortks fine. its just the left " clicky" button thats causing this issue.

mondalaci, 2025-02-01:
Please double-check if you can reproduce this issue with the earlier firmware version and the new firmware version.

Then please state these firmware versions and attach your configuration.

mondalaci, 2025-02-10:
Closing issue due to the lack of feedback.

closed 2025-02-10 by mondalaci