UltimateHackingKeyboard/firmware #1110: "UHK 80 right half" BLE device

mondalaci, 2025-02-01:
See https://forum.ultimatehackingkeyboard.com/t/uhk-80-bluetooth-behavior-when-using-usb-connection-is-this-normal/2023

So far, I haven't been able to make the "UHK 80 right half" reveal itself, and I could use reproduction steps to do so.

In any case, I think this device probably should be made a [directed connectable (ADV_DIRECT_IND)](https://academy.nordicsemi.com/courses/bluetooth-low-energy-fundamentals/lessons/lesson-2-bluetooth-le-advertising/topic/advertising-types/).

benedekkupper, 2025-02-01:
The "UHK 80 right half" is the name of the USB device, Windows UI shows them on the same list without indication of transport.

mondalaci, 2025-02-01:
Ah, I wouldn't have figured this out myself! :)

Given this issue, I'd prefer to include USB vs BLE in device names to increase clarity. Any objections?

benedekkupper, 2025-02-01:
Fine with me