UltimateHackingKeyboard/firmware #1111: UHK80 - Keyboard not waking up

were, 2025-02-02:
https://github.com/user-attachments/assets/f7d8099c-c4e6-4618-bdba-057f33c16c36

I have my UHK80 cable connected to my MacBook.

After bringing MacBook far away and back, I found my UHK80 died.

I press the keys without any reaction --- I'm 95% sure it is NOT a battery issue because use it with cable, which means it is fully charged.

Firmware: 12.3.1

mondalaci, 2025-02-02:
Please push the recessed reset buttons on the bottoms of your UHK halves. Do they make your UHK 80 usable again?

were, 2025-02-02:
> Please push the recessed reset buttons on the bottoms of your UHK halves. Do they make your UHK 80 usable again?

Ok, used the reset button on both sides and it works now.

This message is types through the UHK keyboard.

mondalaci, 2025-02-02:
Glad to hear!

Can you reproduce this issue?

were, 2025-02-03:
Not really, we can close this for now, and I will let you know if this happens again.

closed 2025-02-03 by were