UltimateHackingKeyboard/firmware #1112: Configurable pairing timeout

mondalaci, 2025-02-02:
Based on https://forum.ultimatehackingkeyboard.com/t/uhk-80-switch-off-battery-and-ble/1980/18?u=mlac

Add a new user config variable. I propose a int16 containing the number of minutes, or 0 (pairing disabled) or -1 (pairing allowed forever)

I also propose a start advertising device action that'd use the above variable or 5 minutes if the variable is 0 or -1.

mhantsch, 2025-02-05:
Btw, another strategy that I have seen is that other devices automatically advertise themselves as connectable as long as they are currently not connected to a device. Once they are connected, they stop advertising. To connect them to a new device (and possibly store a new bluetooth connection profile) you have to disconnect them (but not unpair) from the current connection. Or you can press the "make it visible for a short while" button.

mhantsch, 2025-02-05:
> I just wanted to mention that in Windows 11, the UHK80 BLE connection doesn't offer a disconnect option. It only offers the option to remove it entirely. Then you need to restart the Bluetooth radio to pair it again.

In that case you'd have to turn off Bluetooth for a moment on the Windows 11 system to get your device to become disconnected and restart advertising. But that's why a "make visible for a while" button makes sense.

Maybe the point is: as long as the keyboard is not connected at all then it should just keep advertising.

Once it's connected, the advertising stops, and only comes back if the "make visible for pairing" button (= smart macro) is used. (Or becomes disconnected again.)

pcooke9, 2025-02-05:
> In that case you'd have to turn off Bluetooth for a moment on the Windows 11 system to get your device to become disconnected and restart advertising

Pretty annoying if you have multiple devices connected to the same adapter though.
The two devices I have highlighted here are both connected to the same Bluetooth adapter:

[image]

Anyone know if adding a disconnect button is something planned in the future? Sorry if it's been mentioned elsewhere. I don't remember seeing it, if so.

mhantsch, 2025-02-05:
Yeah, I'd set mine to timeout after 5 or 10 minutes, and only turn BT advertisement on when I press some button (to run the appropriate smart macro action).

RyanCwynar, 2025-03-08:
Related to this, in my case, I have two macbooks which I routinely switch between. In order to get to get the UHK to switch between them, I have to tell Macbook 1 to Disconnect. This seems to tell the UHK to stop seeking a connection with that device, which is good. But then Macbook 1 aggressively tries to reconnect and succeeds. So you actually have to turn off the bluetooth radio on Macbook 1 to stop this while the UHK is in the correct state to get it to advertise to Macbook 2. 

Seems like it would be nice to be able to initiate the disconnect and then hold off the aggressive connection attempts from this known connection to allow time to switch to the second without depending on taking actions in the OS settings.

So... macro options to 1. Disconnect current BLE connection. 2. Block traffic from a specific connection

Doesn't sound like good UI for the general user, but seems like something that could be useful macro options.

mondalaci, 2025-03-08:
We could have a restricted and unrestricted advertising mode. In the former, only known devices that are part of host connections can connect to the UHK.

kareltucek, 2025-03-13:
(So far I have implemented the alwaysAdvertiseHid flag. To be continued when we agree on some specification.)

mhantsch, 2025-03-13:
Why do you need the flag? I would use a timeout parameter of 0 to mean "no timeout = always advertise".

kareltucek, 2025-03-13:
You may not want to always advertise.

It is confusing, because that may as well mean don't advertise.

mhantsch, 2025-03-13:
`1` would mean "never", or did you mean `-1` would mean "never"?

kareltucek, 2025-03-13:
I meant advertise for 1 second, meaning advertise almost never.

mhantsch, 2025-03-13:
"almost never", then 😉

kareltucek, 2025-03-13:
(That's why the quotes were there 😃.)

pcooke9, 2025-03-14:
> `1` would mean "never", or did you mean `-1` would mean "never"?

MAX Concise ponders the question of ETERNITY (and fractional divisions thereof)...

mondalaci, 2025-03-15:
Do you guys want to specify the initial pairing timeout, or is a 5-minute hardcoded value sufficient? There'll also be a forever and never option. Please elaborate on your use case if you need a custom pairing timeout.

pcooke9, 2025-03-16:
5 minutes with a never option sounds reasonable to me.

mhantsch, 2025-03-16:
I don't have a particular use case in mind for the *configurability* of the pairing timeout except that I like to be able to adjust things. I would probably set it to even less than 5 minutes, maybe 1 would be a good number for me.