UltimateHackingKeyboard/firmware #1115: Bridge cable still unreliable with 12.3.3

denschub, 2025-02-03:
I updated to 12.3.3 that was released yesterday, and while the crash-loops seem to be gone (at least I couldn't reproduce it anymore), the connection between the two halves somehow got even worse.

Now, I frequently have modifier keys getting stuck, and sometimes a whole second of the left half just not sending any keypresses, and magically recovering itself afterwards. This is making they keyboard near unsuable - but luckily, everything is fine if I just disconnect the cable and use the wireless connection.

I see no log output in Agent, except for a "Dongle and right half run the same dongle protocol version 1.0.0".

Happy to help debug this, if you need anything, do let me know. :)

kareltucek, 2025-02-03:
Thanks for report! 

I have some suspicions and am working on further fixes that relate to the reliability issues -> patience please!

mhantsch, 2025-02-03:
Similar experience here: 12.3.3 is better than the versions before it, but when the bridge cable is connected I still get the odd missing character or garbled text. When the bridge cable is disconnected it seems to work quite well.

Tested it with USB connection to host.

tunney, 2025-02-04:
Yeah I have also found that the bridge cable is quite bad, rather bad actually.

denschub, 2025-02-22:
Welp, no. It just did the thing again where the left half just froze and kept repeating the last character I was pressing endlessly. 

[The OLED showed a `The messanger message pool space ran out!` message](https://github.com/user-attachments/assets/008b731c-dcd7-4d34-b672-fafe0caa5f85), and this was not recoverable until I pressed the reset buttons on both halves.

kareltucek, 2025-03-07:
Regarding the "messanger message pool space" error, please lets continue in https://github.com/UltimateHackingKeyboard/firmware/issues/1144 . 

If there are still other uart troubles on 12.4.0, please open a new ticket ;-).

closed 2025-03-07 by kareltucek