UltimateHackingKeyboard/firmware #1116: 12.3.2 firmware: mass repeating of characters on wake

iamerichb, 2025-02-03:
When the keyboard wakes from sleep (along with the PC), the first character typed after awake will act like it's stuck (not sure if every time yet). Only way I've found to get it to stop is to unplug the USB and hope the keyboard reboots (it eventually does) and then all is well.

This was observed in 12.3.1 and 12.3.2 was supposed to address it.

OS: Windows 11 24H2
Connection: USB to PC, bridge between halves

iamerichb, 2025-02-07:
Happened again today with the 'g' key. PC was asleep overnight, woke up without any keyboard input, 'g' key was first key pressed to wake up the kb and it just repeated endlessly. Unplugged the USB to PC cable, plugged it back in, input stopped but keyboard needed to reboot in order to work again.

mondalaci, 2025-02-09:
How long does it take for your Windows 11 PC to sleep after triggering sleep from the standpoint of USB peripherals? In my experience, depending on the OS, it can take minutes or hours. Knowing the exact duration would help reproduce this issue.

iamerichb, 2025-02-09:
The PC is actually not set to go to sleep (set to Never) nor hibernate.

There is a USB selective suspend setting which *was* enabled and I've just
disabled that to see if it fixes it.

Will update in a few days if it recurs or seems resolved.

iamerichb, 2025-02-11:
New update: today I had the repeating key issue (or another strikingly
similar one) not on the first keystroke of the day.

I started my day: shake mouse to wake monitor up, Windows Hello (IR camera)
authenticated me, clicked on my open Notepad document, hit the 'a' key, no
repeats.. victory!

Went about my business for a couple of minutes, worked in a couple of other
programs, went to an RDP window, hit the 'y' key and repeated until I
unplugged from the right half. Plugged it back in, keyboard was not
registering any typing at all. Unplugged the bridge cable and the keyboard
restarted and all seems to be well.

So, not just 'waking up'.

iamerichb, 2025-03-04:
Any update here? This is still happening. I basically have a text document I keep open for when I come to my computer and start typing in to see if repeating is going to happen.

iamerichb, 2025-03-04:
> We have released version 12.4., which solves some issues compared to
> 12.3.2. We still have some stability problems related to usb suspension and
> to long uptimes.
>
> Anyways, the release is available at
> https://github.com/UltimateHackingKeyboard/firmware/releases

I upgraded to 12.4, and I see this:

[image: image.png]

Keyboard seems to be working fine (USB to right half, bridge cable from
R<>L) but there's an alert in Agent and on the LCD.

iamerichb, 2025-03-04:
I ran the firmware update a second time and this time it reports no issues.

This is the second time updating firmware I've had to run it twice to get
it to not have errors.

kareltucek, 2025-03-04:
The image haven't arrived. If the error is about checksum or dongle protocol version, then you can ignore it given that you are sure that all the devices (dongle, left half, roght half) are flashed to the same firmware version.

iamerichb, 2025-03-06:
The original problem has gotten worse.  After updating to the latest
firmware, now the device doesn't eventually reboot and 'fix' itself.

Now the LCD is stuck on a screen saying 'messenger message pool space ran
out!' and doesn't restart and my keyboard is useless. Since it's got
batteries (which I didn't actually order) I assume I'm going to have to
wait for the batteries to run out before the device will restart unless I'm
missing something? I'm hoping the reset button is not the solution.

I used to unplug the keyboard from USB when the problem happened, sometimes
the bridge cord and shortly thereafter it would reboot the keyboard.  Now,
it's just stuck. I had to change to a different keyboard.

Would really like this fixed or at least back to the previous state when it
would reboot on its own rather than being completely useless.

kareltucek, 2025-03-06:
Yes. You are missing the reset button on the back.

iamerichb, 2025-03-06:
Does the reset button just reboot, or does it reset it back to factory?

kareltucek, 2025-03-06:
Just reboot.

iamerichb, 2025-03-06:
Great, thanks. That will help when this happens again.

Is there a cause found for the repeating issue and if so, any sort of ETA
on a fix?

pcooke9, 2025-03-07:
> I assume I'm going to have to wait for the batteries to run out before the device will restart unless I'm missing something?

You can physically disconnect the batteries if you don't want to use them, btw. Just remove the palm rest pads and plastic inserts, and unplug the battery connectors from the board. Then you can just disconnect the USB cables to restart the UHK80.😉

kareltucek, 2025-03-07:
> Is there a cause found for the repeating issue and if so, any sort of ETA on a fix?

No. We don't know the cause, but are hunting for it. 

Please watch https://github.com/UltimateHackingKeyboard/firmware/issues/1144 for updates, testing firmwares, etc. 

Closing since there is not useful information here atm..