UltimateHackingKeyboard/firmware #1118: UHK 80: only one battery status shown when using Dongle

mhantsch, 2025-02-03:
When I plug the right half into USB, the OLED display shows two battery levels.
When I disconnect USB from the right half, and it connects via the Dongle, then the OLED display only shows one battery level.

It doesn't matter whether I have the two halves connected with the bridge cable or not.

Firmware 12.3.3.

kareltucek, 2025-02-04:
Is this a new regression or an old one? :-)

mhantsch, 2025-02-04:
I don't know, I've never used the Dongle much. Also, I didn't find a previous older issue for it, so I thought I better report it.

mhantsch, 2025-02-15:
With uhk-firmware-12.3.3-uart-fixes-v8.1-bd38000.tar.gz I also saw the single battery when the right half was connected to the host with USB.

closed 2025-02-19 by mondalaci