UltimateHackingKeyboard/firmware #1121: Noise on 12.3.3

tunney, 2025-02-04:
on 12.3.2 if the backlight brightness was set to < 35 or so the noise would be very faint.
On 12.3.3 the backlight brightness setting has no effect and the full noise is always present with the bridge cable in use.

tunney, 2025-02-04:
Reapplying the firmware removed the noise <35 brightness. Strange.

tunney, 2025-02-04:
Interestingly, I left the keyboard on 12.3.3 went to lunch and then the left side (connected with bridge) was totally dead. Reset didn't help, connecting wireless to right half brought it back to live. And then when the bridge cable was reconnected the noise is the loudest it has been.

tunney, 2025-02-04:
[Sounds from a iOS Voice notes recording](https://www.dropbox.com/scl/fi/ngdkygm2rov06d7tnaw6o/keyboard_12.3.3.m4a?rlkey=c0kdodnrx96d1r6zididezo13&st=d5bdwt5q&dl=0)

Nothing fancy, just my phone recording in Voice Notes. The sound (bar the clang) is 100% the left hand side.

tunney, 2025-02-04:
Rolled back to the low baud 12.3.2, 12.3.3 gives me a headache.

pcooke9, 2025-02-04:
> [Sounds from a iOS Voice notes recording](https://www.dropbox.com/scl/fi/ngdkygm2rov06d7tnaw6o/keyboard_12.3.3.m4a?rlkey=c0kdodnrx96d1r6zididezo13&st=d5bdwt5q&dl=0)
> 
> Nothing fancy, just my phone recording in Voice Notes. The sound (bar the clang) is 100% the left hand side.

Well, that's interesting. I can hear the whine just fine in your recording, but I don't hear anything from my UHK80...

Every now & then I can hear a very faint buzz-like sound if I put my ear literally on the keys, but not that high-pitched whine.🤷‍♂️

tunney, 2025-02-04:
And that is with the Key backlight setting at 35........

mondalaci, 2025-02-06:
Firmware 12.3.3 has no related changes and shouldn't affect audible noise. We'll offer a recall to affected customers. I'm closing this issue, but feel free to follow up.

closed 2025-02-06 by mondalaci