UltimateHackingKeyboard/firmware #1124: Disconnecting batteries solved reset issue

Zetaphor, 2025-02-06:
For some time now across firmware versions I have been having an issue where once a day my keyboard would completely lock up and reboot, after which my left side was unsusable. The only solution was to disconnect USB, disconnect the bridge cable, reset both halves, and then reconnect everything.

Unrelated to this I decided to disconnect the batteries in the keyboard as I always use it connected over USB. Interestingly once I did that the daily reset issue has gone away completley. Now the only issue I am encountering is the very occasional key repeat, but it seems to be isolated to the left click of my key cluster, and fairly infrequently. I haven't had any other inputs show the repeating issue.

I'm not sure if this is relevant, but the charging information on the OLED would regularly flip back and forth between showing both sides and only showing the right side.

This can probably be closed, I just wanted to get this documented as it may contain some useful signals.

I am currently running firmware 12.3.3, but this has been an issue pretty much since the first weeks with the prototype board, until I finally disconnected the batteries.

mondalaci, 2025-02-06:
Do you occasionally disconnect your UHK 80 via USB? If so, on such occasions, the UHK 80 gets power-cycled without the batteries, unlike with batteries, resetting firmware state, which may cause the different behavior.

pcooke9, 2025-02-08:
I disconnected the batteries about a week or so ago. I've had the board freeze a couple of times since then, but nowhere near as often as when I had the batteries connected. My UHK80 only really gets power cycled when I shut down the PC for the night or reboot, etc..

I still have the lag/garbled output if the board resets (either intentionally or otherwise), but it doesn't happen very often anymore. But Idk if that's from unplugging the batteries, or firmware v12.3.3.

I guess I'll reconnect them for a while to see if there's a difference.

Zetaphor, 2025-02-08:
> I disconnected the batteries about a week or so ago. I've had the board freeze a couple of times since then, but nowhere near as often as when I had the batteries connected. My UHK80 only really gets power cycled when I shut down the PC for the night or reboot, etc..

Same here, the power is cut off when I shut down the PC, where before the board was staying powered at all times thanks to the batteries. And like @pcooke9 I am seeing significantly less issues as a result. I think I've had to pull the power to reset it once since I made this change, where I was having to do a full reboot/reset at least once every day with the batteries connected.

mondalaci, 2025-02-09:
I don't think a shutdown or reboot power-cycles USB devices, but only cutting power. That is, unplugging and replugging them.

It'd be useful to know what triggers these issues. My guess is sleep-wake events and/or wireless connections (BLE and dongles). This is critical information; without it, we can't reproduce these issues.

Zetaphor, 2025-02-09:
In the situations where I would experience the random lockup requiring a reset I would be in the middle of an hours long session of actively using the keyboard, it was nowhere near the point in which the keyboard had been asleep.

I also don't have the dongle connected and have never connected it to bluetooth.

pcooke9, 2025-02-10:
I'm unsure what triggers it.
I'm also usually actively using the UHK80 around the time it freezes, so it's not sleeping. I don't remember if it's ever frozen while using the dongle/BLE. I normally only use the dongle/BLE for a few minutes after upgrading firmware (to test their basic functions).
___
Regarding the power-cycling...
I have USB Standby Power and Resume by USB Device features disabled in my desktop PC's BIOS.
When I cold-boot my PC with the UHK80's batteries disconnected, I see this on the OLED display:

[image]

I assumed that meant it was being fully power-cycled, as it's similar to what I see when I disconnect/reconnect the USB cable (without batteries) while the PC is powered on.

mondalaci, 2025-02-10:
`agent/packages/usb/get-uptime.ts` provides the uptime of your UHK, such as `0d 11:7:41`, so it's useful to reveal when your UHK was power-cycled the last time.

There's also a more involved way for you guys to help resolve this issue: by connecting a UART adapter:

[image]

You can use your UHK while the adapter interconnects your UHK 80 via UART and your computer via USB. You'd have to run a terminal application on your OS, such as GNU Screen or Minicom, and copy-paste the observed errors. 

Let me know if you're interested, and I can provide it.

pcooke9, 2025-02-10:
> Let me know if you're interested, and I can provide it

Although I'd like to be more helpful, I'm not a developer, and that's all likely far beyond my capabilities.😅

mondalaci, 2025-02-10:
Totally understand! Just to clarify, this would take unscrewing two screws, connecting a cable, and starting an application.

pcooke9, 2025-02-10:
I'm more worried about properly using the software. I've got zero skills behind a terminal, seriously. Especially if I have to work outside of Windows. If it's just a matter of copy/pasting commands or whatever, I can give it a shot. I just have to warn ya, someone might need to walk me through it...

mondalaci, 2025-02-10:
It's just copy-pasting from Minicom on Windows, and I'm glad to help if you get stuck.

pcooke9, 2025-02-10:
Ok. I'm just getting ready for bed now, but you can send me the details on the UHK forum or whatever.🙂

pcooke9, 2025-02-28:
@mondalaci
I got the UART adapter today.

I have no idea what I'm doing, so...
Do I need to install any software/drivers, etc.? Anything I need to know before plugging it in?
I have no idea how to setup & use Minicom either. I have WSL, Ubuntu, and Minicom installed in Windows Terminal, but I really don't know much about Linux at all. Any help would be appreciated.

You might have to explain it like I'm a 5yo.👶🏻😅

mondalaci, 2025-02-28:
Just use Minicom, preferably not via WSL but run it natively on Windows. Choose 115200 baud and the relevant COM port (there shouldn't be many ports). The connect. Any issues, feel free to post a screenshot.

mondalaci, 2025-02-28:
I've just realized that Minicom is not available on Windows natively, and PuTTY would probably be an easier option.

https://pbxbook.com/voip/sputty.html

pcooke9, 2025-02-28:
Ok, I'll try to get it going over the weekend.

pcooke9, 2025-03-04:
Ok, I've been using PuTTY for a few days now.
I haven't had any notable issues to report though. I haven't had any freezes/restarts or output issues at all in quite a long time actually. I've been staying on recent master branch firmware/Agent.

I mostly only see occasional output like the following, which I assume are just minor resets/corrections (I haven't noticed any anomalies in keystroke output behavior when I get these messages):

```
Resetting left right link! Unidirectional
Con state: UartLeft(Disconnected)
Resetting left right link! Bidirectional
Con state: UartLeft(Ready)
Resetting left right link! Bidirectional
Uart: didn't receive ack 6029743, resending
Uart: didn't receive ack 6029884, resending
```
___

Questions-

Are there any docs I can reference to help interpret the PuTTy output, so I don't end up posting random stuff that might not be useful/important?

Does anyone know if there's a way to prefix timestamps to each line of output in PuTTY? It will only add a header with date/time info to the beginning of each session log. I can't seem to get the bell notifications to work, and all the output is contiguous. So I don't know when events actually occur, unless I'm looking at the terminal as it happens.

Are there any specific issues or firmware versions you'd like for me to focus on, or should I just continue using up to date master branch and report whatever?

kareltucek, 2025-03-04:
> Are there any docs I can reference to help interpret the PuTTy output, so I don't end up posting random stuff that might not be useful/important?

No.

> I can't seem to get the bell notifications to work

Not sure what you mean? If a putty feature then I have no advice.

> Does anyone know if there's a way to prefix timestamps to each line of output in PuTTY?

Yes, the general approach is to ask me to do that :-).

> Are there any specific issues or firmware versions you'd like for me to focus on, or should I just continue using up to date master branch and report whatever?

- The "message pool" issue.
- Generally any keyboard freezes.
- Any crashes.

mondalaci, 2025-03-04:
> Does anyone know if there's a way to prefix timestamps to each line of output in PuTTY? It will only add a header with date/time info to the beginning of each session log. I can't seem to get the [...]

On Linux, I use:
`picocom -b 115200 /dev/ttyUSB-right | ts '[%Y-%m-%d %H:%M:%S]' | tee serial_log.txt`

On Windows, [there are some solutions](https://claude.ai/share/37a7592b-3859-460d-9bf7-958b0650b706) according to Claude.

pcooke9, 2025-03-04:
> On Windows, [there are some solutions](https://claude.ai/share/37a7592b-3859-460d-9bf7-958b0650b706) according to Claude.

Thanks! That was the nudge I needed. I ditched PuTTY and tweaked the direct PowerShell option. So, now I have timestamps working in Windows Terminal Preview, which also has custom bell sounds and visual notifications.😁

___

> Not sure what you mean? If a putty feature then I have no advice.

Yeah, I just wanted an audible alert from PuTTY.

> I meant I can add uptime timestamps on firmware side

I suppose that could be useful, but I think I have a workaround for now. (But don't let that stop you if you think it's worth adding.😉)

pcooke9, 2025-03-07:
I guess I'll just ask here, since I don't know where else to post stuff I don't understand...
https://www.youtube.com/watch?v=DU_Gd623HJo&t=19s
___

I got this in the UART adapter log today:

```
[2025-03-06 20:16:43] E: Failed to allocate net_buf 1026
[2025-03-06 20:16:48] E: Failed to allocate net_buf 1026
[2025-03-06 20:17:03] E: Failed to allocate net_buf 1026
[2025-03-06 20:17:08] E: Failed to allocate net_buf 1026
[2025-03-06 20:17:53] E: Failed to allocate net_buf 1026
[2025-03-06 20:17:58] E: Failed to allocate net_buf 1026
[2025-03-06 20:18:23] E: Failed to allocate net_buf 1026
[2025-03-06 20:18:28] E: Failed to allocate net_buf 1026
```

When it started, I think I was just typing and spamming the backspace key a few times. I didn't see any OLED messages, or notice any abnormal behavior while it was happening.

**FW v12.4.0 #efe3a64
Bridge cable & USB both halves to PC
Dongle was plugged into PC, but disabled via** `set bluetooth.enabled 0`.

kareltucek, 2025-03-07:
> I guess I'll just ask here, since I don't know where else to post stuff I don't understand...

Either a new ticket, or just shoot me a quick pm as to whether or not it is interesting. 

-> https://github.com/UltimateHackingKeyboard/firmware/issues/1169