UltimateHackingKeyboard/firmware #1131: Mouse speed faster when connected via BT

tunney, 2025-02-14:
When a UHK80 is connected to the host via BT the mouse keys result in the mouse moving much faster than when a UHK80 is connected wired to a host.

(I would post a video of the difference but the hassle of changing connections is still around)

tunney, 2025-03-05:
Adding video 

https://www.dropbox.com/scl/fi/acy9ape04p2kqnih6s3pl/mouseSpeed_wired_usb_difference.mp4?rlkey=bw006l35rrqpnu0fipr4028zf&st=y26kyvmw&dl=0

mondalaci, 2025-03-05:
BLE mouse key speed is consistently 3x faster than USB mouse key speed on macOS and Windows, but not Linux, where they're equal.

Module pointer speed, however, is the same on every OS via BLE and USB.

@kareltucek Any insights?

kareltucek, 2026-11-05:
Hypothesis: ble is "reliable", so resends will apply multiple times. Current resend timeout is 32ms though, which should be plenty and definitely not cause 3x faster.