UltimateHackingKeyboard/firmware #1133: Issue  in UHK 80 while connecting over dongle

mohitt, 2025-02-15:
This happens only with firmware 12.3.3 onwards and my usb dongle connection. 

My keyboard stops taking any input and if I start the `UHK Agent` and connect over USB I see following messages 

```
D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>>> Messanger message pool space ran out!

D>
```

[image]

mohitt, 2025-02-16:
It is annoying that keyboard stops working and no way to fix it but wait. For it to go standby and power out

kareltucek, 2025-02-16:
Sorry! Yes, I know about it. 

Powercycling the right half or the dongle should help temporarily.

As for a proper fix, it would help if we can figure out a way to reproduce it, out at least figure out the conditions when it tends to happen. 

(Also, always make sure your dongle is on the exactly same firmware as the right half.)

mohitt, 2025-02-16:
@kareltucek how should I power cycle only way I know is to let it sleep and leave it for a few minutes. but, that is really annoying.

mondalaci, 2025-02-16:
With the recessed reset button on the bottom.

mohitt, 2025-02-16:
Got it will try next time, as of no i downgraded and started using usb
wired connection only

rkoopmann, 2025-02-18:
I'm getting the same error with direct & indirect connection of each half (with and without bridge cable). I have a wired connection from UHK80 to MBP M3.

Possibly related, in uhkagent, I'm seeing the touchpad plugin pop in briefly before disappearing. I've never connected a touchpad module to this keyboard. here's a gif [image]

After a day or so, the display shows a triangle shape and the keyboard eventually becomes unresponsive; Reflashing the firmware seems to clear message pool space, but the phantom touchpad module remains an issue. After a day or so, same thing happens.

pcooke9, 2025-02-18:
> I'm seeing the touchpad plugin pop in briefly before disappearing

This is fixed with: https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v12.3.4

kareltucek, 2025-05-28:
`Messanger message pool space ran out!` should no longer be a problem either!