UltimateHackingKeyboard/firmware #1141: Make NUS use directed advertising

mondalaci, 2025-02-20:


kareltucek, 2025-02-21:
@mondalaci *why* exactly do you want this? Just to hide the nus advertisement from other devices?

If so, then this doesn't achieve the goal, and the following excerpt explains why (assuming its accuracy):

[image]

mondalaci, 2025-02-21:
I may be on the wrong path, but the objective is to make NUS advertisements invisible to third-party devices.

@benedekkupper Please provide more insight.

benedekkupper, 2025-02-21:
The directed advertising is what we tried to make work in #1122 , but didn't stick, and was disabled. Directed advertising is a thing though, and Zephyr has support for it, even a demo sample to test: https://docs.zephyrproject.org/latest/samples/bluetooth/direct_adv/README.html, so maybe it's a matter of figuring out what BT address transformation is necessary to get it to work on the UHK.

mondalaci, 2025-02-21:
@benedekkupper Please let me know how to move forward with this. Maybe you could add the relevant code (but not remove a ton of needed code :) and Karel would take over.

kareltucek, 2025-02-21:
@benedekkupper the code from #1122 is still in place, just refactored so that you have a single site that provides you with everything necessary for further experimenting. If you can fix the code, please do so: https://github.com/UltimateHackingKeyboard/firmware/blob/5d4773e7835b6f5fb32a0f4f4a58e1d577b29ebf/device/src/bt_advertise.c#L81-L90 

I have a feeling like we are mixing something like four different features together:
1) Single-target directed advertisement, which is what @benedekkupper talks about, in which the peripheral sends advertisement to a single central.
2) Round-robin-switched directed advertisement which would leverage the above option so that a list of addresses can be targeted. I have no idea whether zephyr has a built-in support for it, or whether we would have to implement it ourselves. (At least in the latter, getting the above to work is a prerequisite.)
3) A directed advertising that would broadcast a single packet to a list of ble addresses such that only those devices could see. This is what @mondalaci wants, and what probably doesn't exist.
4) Scan response filtering thingy, which first publicly broadcasts advertising, but allows specifying a list of addresses that the peripheral is willing to send additional data if the centrals (from the list) ask for it. This is what @mondalaci has provided a link to. However I don't think it solves any of our issues since the advertisement is publicly broadcasted and so any bluetooth search will show the device in the connection lists.

In any case, I consider this entire request to be an exotic aesthetic detail that I hardly should be wasting my time on these days - since we have a long backlog of issues that would bring much more benefit to the users than this.

(@benedekkupper in case you want to try to solve the "directed broadcast" scenario, then let me know - I will share a branch that is prepared for it...)