UltimateHackingKeyboard/firmware #1143: KVM Hotkeys Not Working With UHK80

breser, 2025-02-22:
I have a UHK60v1 and a UHK80.  Both of them are on the (at this time) current firmware of 12.3.3.

I also have a Level1Techs DP 1.4 KVM:
https://www.store.level1techs.com/products/p/14-kvm-switch-dual-monitor-2computer-z5erd-n6mbj

With the UHK60v1 I am able to use the hotkeys the KVM supports to switch devices.

With the UHK80 I am not.

I've seen this issue from the past:
https://github.com/UltimateHackingKeyboard/firmware/issues/536

But it seems that for the people that needed just 6KRO to have it work properly with their KVM that issue has not been resolved in the current firmware.  However, I do not believe my problem requires 6KRO because it's working with my UHK60v1 on the current versions.

Please note that I'm using USB with both devices and both devices are plugged into the proper HID ports on the KVM that are required for this to work.

Given some of the discussion in the older issue though, I wonder if there isn't some sort of USB ordering issue that might be relevant in this case.

If there are ideas on things to try I'm happy to give them a try.

mondalaci, 2025-02-22:
The UHK 60 and UHK 80 use different USB firmware stacks, and the USB descriptors/reports probably differ somewhat, which is why you experienced a difference.

Tailoring the descriptors to make them work with most KVM switches can be very time-consuming. I think it's a better strategy to [implement 6KRO](https://github.com/UltimateHackingKeyboard/firmware/issues/536) instead.

@benedekkupper Feel free to provide more insight.

benedekkupper, 2025-02-22:
Well we do have the 6KRO fallback option in place on UHK80 (`HID_SetKeyboardRollover`), the only thing that's missing is a user configuration mapping AFAIK, so that it can be controlled from the agent, and it keeps its value through power cycles. I can prepare a test FW to verify if the dynamic 6KRO selection is functional with the KVM, if yes, then we can go forward with the configuration changes.

mondalaci, 2025-02-22:
1. Can we also fall back to 6KRO on the UHK 60?
2. Does this fallback work at runtime with both the UHK 60 and UHK 80?

benedekkupper, 2025-02-22:
This fallback isn't done on the UHK60 yet.

mondalaci, 2025-02-22:
How much work is it?

benedekkupper, 2025-02-25:
Here is an UHK80 firmware that uses 6KRO: https://github.com/UltimateHackingKeyboard/firmware/actions/runs/13520273276/artifacts/2648024398

breser, 2025-02-25:
Flashed that firmware on my UHK80 and it did not help with the KVM hotkeys for me.  I'm not entirely surprised as the UHK60v1 has firmware that has NKRO and it works with the KVM.  So I don't think the problem is something else about how USB is registered.

I spent some time over the weekend trying to get the firmware building locally and was having some weird problems (not going to bring them up here because it's probably just me).  But once I work that out I'll poke around more and see what I can figure out.  If you have any suggestions I'm open to the suggestions.

mondalaci, 2025-02-25:
@benedekkupper What can be the culprit? The implementation of the descriptors?

breser, 2025-02-25:
For what it's worth I don't really care if the agent or the update process works over the HID ports.  It's not a big deal to me to have to plug it in directly to a computer for that.  I rarely use the agent or update the firmware.  So fixing that IMHO is not in the scope of what I'm trying to get fixed here.

mondalaci, 2025-02-25:
Indeed, I failed to catch this detail!

@kareltucek If you have time, please look into the report sending logic of #1148.

kareltucek, 2025-02-25:
@mondalaci / @breser if you think this is a report logic issue, then please elaborate on how the scancodes sent by uhk80 differ from those sent by uhk60.

(Or maybe I don't understand what kvm hotkeys are? I assume they are normal key combos that the kvm intercepts and interprets as is own controls.)

mondalaci, 2025-02-25:
@kareltucek Your assumption is correct. I don't have anything else to add.

mhantsch, 2025-02-25:
@breser Have you tried increasing the Keystroke delay on the UHK?

breser, 2025-02-26:
So I'm not so sure it's not a USB registration issue because from my understanding is the KVM will just pass through devices that don't register themselves properly as keyboards and just ignore the data for them.  Devices it recognizes as USB HID keyboards it will listen for the events looking for the hotkeys.  So if the UHK80 doesn't identify itself in a way the KVM recognizes as a keyboard then it could still work with the host while the KVM hotkeys are not registered.

The KVM hotkeys are documented here (note only the KVM ones apply, the KM ones are for something different than what I have):
https://forum.level1techs.com/t/official-l1techs-products-technical-information-wip/204530

But the general idea is that you double tap a leading key followed by some keys to issue a command.  The leading keys can be Left-Control, Right-Control, Scroll Lock or Num-Lock, but Left-Control and Scroll Lock are enabled by default.  For example you could double tap Left-Control and then 1 to switch to the first computer and double tap Left-Control and then 2 to switch to the second computer.

Using KeyCodes on the Mac I see the following for the double tap Left-Control and then 1 with the UHK80:
Modifier Change
	Keys:		⌃
	Modifiers:	262401 / 0x40101 

Modifier Change
	Modifiers:	256 / 0x100 

Modifier Change
	Keys:		⌃
	Modifiers:	262401 / 0x40101 

Modifier Change
	Modifiers:	256 / 0x100 

Key Down
	Characters:	1
	Unicode:		49 / 0x31
	Keys:		1
	Key Code:	18 / 0x12
	Modifiers:	256 / 0x100 

Key Up
	Characters:	1
	Unicode:		49 / 0x31
	Keys:		1
	Key Code:	18 / 0x12
	Modifiers:	256 / 0x100 

I see the same from the UHK 60, except that when it's plugged into the HID port on the KVM the Mac does not see the Key Down or Key Up modifiers for the 1 key since it appears that the KVM consumes it and doesn't pass it along.

Both keyboards have the Keystroke delay set at the default of 0ms.  But I've tried every 5 ms all the way up to 50ms on the UHK80 with no difference in behavior.  The UHK60v1 works at 0ms or 50ms just fine (didn't bother to try everything between since it didn't seem useful given that it works at the extremes).

In the Wooting thread you can see [Wendel say that the keyboard needs to be HID 1.1 compliant and ask if the keyboard is using PHY or bit-banging as apparently bit-banging keyboard implementations don't work](https://forum.level1techs.com/t/wooting-60he-and-hid-ports/210911/27).  I wonder if he doesn't mean HID 1.11 compliant since I see 1.11 is the common referenced HID specification.

It's possible the Key Codes app I was using on the Mac is not low level enough to see any potential difference.  I'll try to find some more time to get out a Linux machine to put behind this KVM setup and see if I can run xev on it and see if there are any differences there.  I'll also try to figure out how to sniff the USB registration of both devices to look for a difference while I'm at it.

Happy to investigate anything else that you can suggest.

kareltucek, 2025-02-26:
> It's possible the Key Codes app I was using on the Mac is not low level enough to see any potential difference. 

No need to.

If I understand correctly, you are using simple key actions for both left control as well as the numbers (i.e., not secondary roles, not macros, not actions that combine modifier with a scancode).

If that is the case, It might be worth to check that left control is really left control and not right. 

Apart from that I would rule out the high level logic.

------

Have you checked the number of exposed interfaces? I think that gamepad should be disabled by default, but haven't checked.

breser, 2025-02-26:
Correct I've been trying to use just manual key strokes to do the hotkey.  I'm using a very slightly modified keymaps based on he QWERTY Mac and PC keymaps.  Both of them have the LCtrl as the Modifier for the bottom left key which is labeled Ctrl.  I had to add ScrLk binding back to the Mac version of the keybinding since it isn't turned on there and it has the Scancode selected as Scroll Lock.  Neither key has any other secondary roles or anything else going on with it.

How do I go about checking the number of exposed interfaces?  I know I saw some GamePad stuff looking around the firmware but I don't see anything in the Agent about it.  I do have a Key cluster module and all the navigation modes are set to Scroll, Caret or Cursor.

Also worth mentioning that the KVM doesn't recognize the key strokes even when the Key Cluster is not installed on the UHK80.  I didn't try the UHK60v1 with the Key Cluster.  But I don't think it's really relevant, I'm not using any of the keys for this and it doesn't seem to make a difference with it on or off.

kareltucek, 2025-02-26:
Modules are not relevant.

As for interfaces, I guess this should be a good starter: https://chatgpt.com/share/67bf6765-09a8-8005-8d91-3b0e66dd7678

breser, 2025-02-26:
No output from the `ioreg` command has the word keyboard in it.  But here's the output of `ioreg -p IOUSB -w0` with everything irrelevant removed:
```
+-o Root  <class IORegistryEntry, id 0x100000100, retain 35>
  +-o AppleT8112USBXHCI@02000000  <class AppleT8112USBXHCI, id 0x10000049d, registered, matched, active, busy 0 (12 ms), retain 37>
  +-o AppleT8112USBXHCI@00000000  <class AppleT8112USBXHCI, id 0x1000004d3, registered, matched, active, busy 0 (11880 ms), retain 598>
  | +-o TS4 USB2.0 Hub@00100000  <class IOUSBHostDevice, id 0x100062c8a, registered, matched, active, busy 0 (3175 ms), retain 38>
  | | +-o TPS DMC Family@00130000  <class IOUSBHostDevice, id 0x100062cae, registered, matched, active, busy 0 (4 ms), retain 17>
  | | +-o USB2.1 Hub@00150000  <class IOUSBHostDevice, id 0x100062cb0, registered, matched, active, busy 0 (2835 ms), retain 32>
  | | | +-o Virtual HUB@00153000  <class IOUSBHostDevice, id 0x100062cd4, registered, matched, active, busy 0 (490 ms), retain 41>
  | | |   +-o Razer DeathAdder V3@00153200  <class IOUSBHostDevice, id 0x100062d3e, registered, matched, active, busy 0 (57 ms), retain 39>
  | | |   +-o UHK 80 right half@00153100  <class IOUSBHostDevice, id 0x1000d3715, registered, matched, active, busy 0 (99 ms), retain 51>
  | | |   +-o Virtual HID@00153600  <class IOUSBHostDevice, id 0x1000d394a, registered, matched, active, busy 0 (58 ms), retain 33>
  | | +-o TS4 USB2.0 HUB@00110000  <class IOUSBHostDevice, id 0x100062d08, registered, matched, active, busy 0 (207 ms), retain 36>
  | | | +-o IOUSBHostDevice@00115000  <class IOUSBHostDevice, id 0x100062dd1, registered, matched, active, busy 0 (22 ms), retain 22>
  | | | +-o TS4 USB2.0 HUB@00112000  <class IOUSBHostDevice, id 0x100062e0b, registered, matched, active, busy 0 (17 ms), retain 33>
  | | | +-o TS4 USB2.0 HUB@00111000  <class IOUSBHostDevice, id 0x100062e3f, registered, matched, active, busy 0 (140 ms), retain 31>
  | | +-o USB-C To DP Cable@00140000  <class IOUSBHostDevice, id 0x1000d3b61, registered, matched, active, busy 0 (19 ms), retain 18>
  | +-o USB3.0 Hub@00200000  <class IOUSBHostDevice, id 0x100062c9d, registered, matched, active, busy 0 (1327 ms), retain 34>
  |   +-o USB3.2 Hub@00210000  <class IOUSBHostDevice, id 0x100062cb5, registered, matched, active, busy 0 (1258 ms), retain 31>
  |   +-o TS4 USB3.2 Gen2 HUB@00240000  <class IOUSBHostDevice, id 0x100062d0a, registered, matched, active, busy 0 (55 ms), retain 33>
  |     +-o TS4 USB3.2 Gen2 HUB@00242000  <class IOUSBHostDevice, id 0x100062e0e, registered, matched, active, busy 0 (15 ms), retain 31>
  |     +-o TS4 USB3.2 Gen2 HUB@00241000  <class IOUSBHostDevice, id 0x100062e15, registered, matched, active, busy 0 (19 ms), retain 31>
  +-o AppleT8112USBXHCI@01000000  <class AppleT8112USBXHCI, id 0x100000596, registered, matched, active, busy 0 (12 ms), retain 37>
```

All the TS4 stuff is the Thunderbolt 4 dock that the computer is attached to.  USB-C to DP Cable is self explanatory.  Razer thing is my mouse. Most of the HUBs and crap are from the Thunderbolt4 dock.  Virtual HUB the hub on the KVM for the HID ports, Virtual HID is related to the KVM HID support.  And the Apple stuff is just the built in busses for each USB-C port on the Mac.

system_profiler command again doesn't have anything with the word keyboard in it but here's the output which is really just more detail about individual devices on the USB busses:
```
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT8112USBXHCI

    USB 3.1 Bus:

      Host Controller Driver: AppleT8112USBXHCI

    USB 3.1 Bus:

      Host Controller Driver: AppleT8112USBXHCI

        USB3.0 Hub:

          Product ID: 0x0b40
          Vendor ID: 0x8087  (Intel Corporation)
          Version: 12.34
          Speed: Up to 10 Gb/s
          Manufacturer: Intel Corporation.
          Location ID: 0x00200000 / 2
          Current Available (mA): 900
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            TS4 USB3.2 Gen2 HUB:

              Product ID: 0x5500
              Vendor ID: 0x2188  (CalDigit)
              Version: 1.02
              Speed: Up to 10 Gb/s
              Manufacturer: CalDigit, Inc
              Location ID: 0x00240000 / 10
              Current Available (mA): 900
              Current Required (mA): 0
              Extra Operating Current (mA): 0

                TS4 USB3.2 Gen2 HUB:

                  Product ID: 0x5502
[... 854 lines cut ...]
0x0      0x0       0x0        65280     72    0x1000009a1 SPMI      AppleBTM                   BTM                                (null)                    0
0x5ac    0x8104    0x0        65280     5     0x100000840 SPU       AppleSPUHIDDevice          (null)                             (null)                    1
0x37a8   0x1       0x153100   1         2     0x1000d4788 USB       AppleUserHIDDevice         UHK 60 v1                          AppleUserUSBHostHIDDevice 0
0x0      0x0       0x2d       65280     11    0x100000ad1 FIFO      AppleHIDTransportHIDDevice Apple Internal Keyboard / Trackpad (null)                    1
0x1532   0xb2      0x153200   1         6     0x100062d54 USB       AppleUserHIDDevice         Razer DeathAdder V3                AppleUserUSBHostHIDDevice 0
0x0      0x0       0x0        65292     1     0x100000844 SPU       AppleSPUHIDDevice          (null)                             (null)                    1
[... cut ...]
0x37a8   0x1       0x153100   12        1     0x1000d4786 USB       AppleUserHIDDevice         UHK 60 v1                          AppleUserUSBHostHIDDevice 0
0x0      0x0       0x2d       65280     95    0x100000ac9 FIFO      AppleHIDTransportHIDDevice Apple Internal Keyboard / Trackpad (null)                    1
```

UHK60v1 shows up as 1 more device and the same number of services.

Note I did plug the KVM USB directly into the laptop rather than through the Thunderbolt 4 dock.  That doesn't change anything with respect to the KVM hotkeys working or not, UHK80 still does not and UHK60v1 still does.

pcooke9, 2025-02-26:
> Have you checked the number of exposed interfaces? I think that gamepad should be disabled by default, but haven't checked.

I don't know if this is helpful at all, but I just wanted to mention the UHK80 does indeed install gamepad components in Windows. The UHK80 auto-installs an **"Xbox 360 Wireless Receiver for Windows"** for each half, as well as the dongle.

[image]

I was having weird shutdown/restart hangs on Win 11 a while back, and I discovered it was caused by the UHK80's "Xbox 360 Wireless Receiver for Windows". The hang issue has since been resolved, but while troubleshooting the issue I saw this: https://github.com/UltimateHackingKeyboard/firmware/issues/770#issuecomment-2070817465

So I assumed a stray registry entry from the UHK60 was causing the "Xbox 360 Wireless Receiver for Windows" to be erroneously auto-installed with the UHK80, but it also gets installed when connecting the UHK80 for the first time on a fresh Windows install (no prior UHK60 or Xbox devices connected).

breser, 2025-02-26:
Does the gamepad thing happen with USB or just wireless?  Because I'm just using USB?

pcooke9, 2025-02-27:
> Does the gamepad thing happen with USB or just wireless? Because I'm just using USB?

The "Xbox 360 Wireless Receiver for Windows" is added when connecting either half to the PC via USB, as well as when the dongle is connected to the same PC via USB. It doesn't get installed when I connect the UHK80 to the same PC via BLE using an ASUS USB-BT500 adapter.

breser, 2025-03-03:
Put that firmware on my UHK80 and it still does not trigger KVM hotkeys.

Here's the UHK60 data from USB Device Tree Viewer v4.4.4 on Windows:
```
    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-1-3-1
Properties               : 0x00
 IsUserConnectable       : no
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x01 (Port 1)

      ========================== Summary =========================
Vendor ID                : 0x37A8 (Ultimate Gadget Laboratories Kft.)
Product ID               : 0x0001
Manufacturer String      : Ultimate Gadget Laboratories
Product String           : UHK 60 v1
Serial                   : 2118271812
USB Version              : 2.0.1
Port maximum Speed       : Full-Speed
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 100 mA
Used Endpoints           : 7

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_37A8&PID_0001#2118271812#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-23
Device ID                : USB\VID_37A8&PID_0001\2118271812
Hardware IDs             : USB\VID_37A8&PID_0001&REV_0001 USB\VID_37A8&PID_0001
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0047 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.26100.1882  Date: 2025-02-11  Company: Microsoft Corporation)
Driver Inf               : C:\Windows\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0009
Address                  : 1
Location IDs             : PCIROOT(0)#PCI(0801)#PCI(0004)#USBROOT(0)#USB(1)#USB(3)#USB(1), ACPI(_SB_)#ACPI(PCI0)#ACPI(GP17)#ACPI(XHC1)#ACPI(RHUB)#ACPI(PRT1)#USB(3)#USB(1)
Container ID             : {7691a097-2e3f-58c7-95fc-5355face13c4}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180400A (DN_DRIVER_LOADED, DN_STARTED, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
EnhancedPowerMgmtEnabled : 0
Power State              : D0 (supported: D0, D1, D2, D3, wake from D0, wake from D1, wake from D2)

        +++++++++++++++++ Registry USB Flags +++++++++++++++++
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\37A800010001
 MsOs20DescriptorSetInfo : REG_BINARY 00 00 03 06 00 00 01 01

        ---------------- Connection Information ---------------
Connection Index         : 0x01 (Port 1)
Connection Status        : 0x01 (DeviceConnected)
Current Config Value     : 0x01 (Configuration 1)
Device Address           : 0x03 (3)
Is Hub                   : 0x00 (no)
Device Bus Speed         : 0x01 (Full-Speed)
Number of open Pipes     : 0x06 (6 pipes to data endpoints)
Pipe[0]                  : EndpointID=1  Direction=IN   ScheduleOffset=0  Type=Interrupt  wMaxPacketSize=0x1D    bInterval=1   -> 392 Bits/ms = 49000 Bytes/s
[... 1266 lines cut ...]
Enumerator               : HID
Location Info            : -
Address                  : 1
Manufacturer Info        : Microsoft
Capabilities             : 0xA0 (SilentInstall, SurpriseRemovalOK)
Status                   : 0x0180200A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
EnhancedPowerMgmtEnabled : 0
Power State              : D0 (supported: D0, D1, D2, D3, wake from D0, wake from D1, wake from D2)

             +++++++++++++ Mouse Information ++++++++++++++
Input Data Queue Length  : 2
Mouse Identifier         : 256
Number of Buttons        : 20
Sample Rate              : 0

             ++++++++++++++ HID Information +++++++++++++++
Manufacturer             : Ultimate Gadget Laboratories
Product                  : UHK 80 Right USB
Serial Number            : 4011737106
UsagePage                : 0x01 (Generic Desktop Controls)
Usage                    : 0x02 (Mouse)
```

So looks like the gamepad support is not the source of my problem.  There are still the errors reading the HID descriptor which happens only on the UHK80, but I'm not sure what the reason for that is.  Like I said previously the F1 help document that it says to look at doesn't provide any insight into the error.

breser, 2025-09-17:
I haven't had much time to dig into this.  I gave 15.0.1 a spin today and the KVM hotkeys still do not work.

But I did see an update form the folks that make the KVM.  They said that in one keyboard firmware case that ["the hid device has to be one of the first four endpoints."](https://forum.level1techs.com/t/uhk80-and-kvm-hotkeys/226330/6?u=breser)  I've looked through my previous USB dumps but I'm not sure if I can tell if the HID device is in the first 4 endpoints.

If there's a firmware to try I'm happy to give it a spin if the above gives an idea on how to fix this.

mondalaci, 2025-09-17:
@benedekkupper Is the UHK 80 keyboard included in the first 4 endpoints? If not, is it planned? If not, why?

breser, 2025-09-25:
As I said in my initial report, this isn't an issue for the UHK60.  So I don't know what you mean by resolving the issue for both.

UHK60v1 (with 15.0.1 firmware): KVM Hotkeys work
UHK80 (with 15.0.1 firmware and 15.2.0#771e5e4 firmware): KVM Hotkeys do NOT WORK.

I do not have any special settings on my UHK60v1 with this working.  In fact at one point after getting the UHK60v1 I bricked it and had to reset it and since I haven't been using it I never got around to putting my settings back on it.  So it's entirely stock.

If your intent is to use the UHK80 USB implementation on the UHK60 then that may very well break it for both.

mondalaci, 2025-09-25:
The 6KRO mode of c2usb should resolve this issue for both models. Every other solution is potentially very time-consuming and doesn't fully solve the issue for a wide array of KVMs.