UltimateHackingKeyboard/firmware #1145: UHK v80, USB connection doesn't wake up After suspend

mohitt, 2025-02-23:
My UHK v80 would connect even on USB connection once it has gone past the sleep time. I have to push the reset button the back to get the usb connection to work.

mondalaci, 2025-02-23:
What's your OS?

Related issues: #1098, #1087, #1060, #1017.

I've captured this log on Windows 11 with firmware 12.3.5. I connected the UHK 80 to the laptop at 23:49 and woke it up at 10:04. I tried to press the keys to no avail. The display shows disconnected. Replugging USB doesn't work, and "USB device is not recognized" is displayed by Windows eventually. Only resetting the right half fixes the issue.

```
[2025-02-22 23:49:16] Con state: Host0(My PC, Disconnected), Active
[2025-02-22 23:49:16] Supplied connection (0) doesn't correspond to a host connection!
[2025-02-22 23:49:16] Con state: Host0(My PC, Ready)
[2025-02-22 23:49:16] Conn state: Host0(My PC, Ready), Active
[2025-02-23 02:35:35] Con state: Host0(My PC, Disconnected), Active
[2025-02-23 02:35:35] Supplied connection (0) doesn't correspond to a host connection!
[2025-02-23 02:35:35] Con state: Host0(My PC, Ready)
[2025-02-23 02:35:35] Conn state: Host0(My PC, Ready), Active
[2025-02-23 02:35:48] Con state: Host0(My PC, Disconnected), Active
[2025-02-23 02:35:48] Supplied connection (0) doesn't correspond to a host connection!
[2025-02-23 10:04:08] Con state: Host0(My PC, Ready)
[2025-02-23 10:04:08] Conn state: Host0(My PC, Ready), Active
[2025-02-23 10:04:14] Con state: Host0(My PC, Disconnected), Active
[2025-02-23 10:04:14] Supplied connection (0) doesn't correspond to a host connection!
[2025-02-23 10:04:14] E: Failed to allocate net_buf 64
[2025-02-23 10:04:15] E: Failed to allocate net_buf 64
[2025-02-23 10:04:17] USB: requesting remote wakeup
[2025-02-23 10:04:17] USB: remote wakeup disabled
[2025-02-23 10:04:17] Can't send report - selected connection is not ready!
[2025-02-23 10:04:17] Giving report to c2usb anyways!
[2025-02-23 10:04:17] Can't send report - selected connection is not ready!
[2025-02-23 10:04:17] Giving report to c2usb anyways!
[2025-02-23 10:04:20] E: Failed to allocate net_buf 64
[2025-02-23 10:04:21] Can't send report - selected connection is not ready!
[2025-02-23 10:04:21] Giving report to c2usb anyways!
[2025-02-23 10:04:21] Can't send report - selected connection is not ready!
[2025-02-23 10:04:21] Giving report to c2usb anyways!
[2025-02-23 10:04:26] E: Failed to allocate net_buf 64
[2025-02-23 10:04:31] E: Failed to allocate net_buf 64
[2025-02-23 10:04:37] E: Failed to allocate net_buf 64
[2025-02-23 10:04:43] E: Failed to allocate net_buf 64
[2025-02-23 10:04:48] E: Failed to allocate net_buf 64
[2025-02-23 10:04:56] E: Failed to allocate net_buf 64
[2025-02-23 10:05:02] E: Failed to allocate net_buf 64
[2025-02-23 10:05:05] USB: requesting remote wakeup
[2025-02-23 10:05:05] USB: remote wakeup disabled
[2025-02-23 10:05:05] Can't send report - selected connection is not ready!
[2025-02-23 10:05:05] Giving report to c2usb anyways!
[2025-02-23 10:05:05] Can't send report - selected connection is not ready!
[2025-02-23 10:05:05] Giving report to c2usb anyways!
[2025-02-23 10:05:05] Can't send report - selected connection is not ready!
[2025-02-23 10:05:05] Giving report to c2usb anyways!
[2025-02-23 10:05:05] Can't send report - selected connection is not ready!
[2025-02-23 10:05:05] Giving report to c2usb anyways!
[2025-02-23 10:05:07] Can't send report - selected connection is not ready!
[2025-02-23 10:05:07] Giving report to c2usb anyways!
[2025-02-23 10:05:07] Can't send report - selected connection is not ready!
[2025-02-23 10:05:07] Giving report to c2usb anyways!
[2025-02-23 10:05:07] E: Failed to allocate net_buf 64
[2025-02-23 10:05:13] E: Failed to allocate net_buf 64
[2025-02-23 10:05:46] USB: requesting remote wakeup
[2025-02-23 10:05:46] USB: remote wakeup disabled
[2025-02-23 10:05:46] Can't send report - selected connection is not ready!
[2025-02-23 10:05:46] Giving report to c2usb anyways!
[2025-02-23 10:05:47] Can't send report - selected connection is not ready!
[2025-02-23 10:05:47] Giving report to c2usb anyways!
```

Can't send report - selected connection is not ready!
Giving report to c2usb anyways!
Can't send report - selected connection is not ready!
Giving report to c2usb anyways!
Can't send report - selected connection is not ready!
Giving report to c2usb anyways!

mohitt, 2025-02-23:
My os is windows 11 as well, and yes resetting the right is the only way to fix it. Everyday I am resetting the right half like 3-4 times. Everytime I come back from a break pretty much

mondalaci, 2025-02-24:
Assuming it's a regression, would you narrow it down to the exact firmware release that introduces this issue, starting from firmware 12.0.0?

mohitt, 2025-02-27:
This is annoying in a day, I have to do this atleast 4-5 times. Now, I tried ble and dongle connection. On BLE and dongle connection my keyboard macros are not typing the keys properly

 A $500 keyboard. with so many issues. I have used v1 and v2 both. Both, were great. So, much so. That today I stopped using UHK v80. Since it is so annoying to use.

mondalaci, 2025-02-27:
Let me find the problematic firmware version and drill down to the exact commit. It'll probably take a few days.

mohitt, 2025-02-27:
Thanks @mondalaci but, why does it drop characters that are supposed to be type as part of the macro in case of Dongle or BLE ?

mondalaci, 2025-02-27:
An old bug caused dropped characters, but it was fixed a while back. I can look into it if you share your configuration. Tell me which macro to trigger.

mohitt, 2025-02-27:
Cant share config as is, It has some sensitive data as well. I can remove the sensitive data then share

mondalaci, 2025-02-27:
Please do.

mohitt, 2025-02-27:
[UserConfiguration.json](https://github.com/user-attachments/files/19016419/UserConfiguration.json)

mondalaci, 2025-02-27:
Which macros drop characters via BLE or the dongle? Please provide some examples.

mohitt, 2025-02-28:
as simple as `typewin` is dropping characters.

mondalaci, 2025-02-28:
I tried the `typewin` macro about 100 times via the dongle and BLE, and `Blah##@@` was consistently typed correctly every occasion.

Which firmware version are you running? Do you run any software that might interfere with keyboard input?

mondalaci, 2025-02-28:
Synergy is the likely cause. Does this issue occur only with the dongle and BLE but not USB? I think it should occur with USB, too.

To combat this issue, please create a `$onInit`-named macro that includes the `set keystrokeDelay 10` macro command and adjust the delay value if necessary.

kareltucek, 2025-02-28:
Iirc keystroke delays can these days be set via gui under typing behavior too ;-).

mondalaci, 2025-02-28:
Indeed! I'm so behind the curve with these things! :)

mohitt, 2025-02-28:
It is not happening on USB, for sure. I can try disabling synergy and try after that 
and sure I can set keystroke delay as well. Do we have any update on keyboard dropping the USB connection, after standby ?

mondalaci, 2025-02-28:
You're welcome to test the mentioned measures.

The original issue is not resolved yet, and fixing it'll take longer.

mohitt, 2025-03-02:
So, my analysis. Tells me if the computer doesn't go to sleep, the connection is not dropped. If the computer goes to sleep the connection is dropped.

kareltucek, 2025-03-06:
@benedekkupper I have captured this log with 3566bfc67f59b2f424052d8e17958e7fef2f341b (current master approx 12.4.0) against a windows 11 machine:

```
// connected keyboard:

uhk80:right$ =============
L0 ON
=============
Con state: Host0(Usb, Ready)
Conn state: Host0(Usb, Ready), Active
Conn state: Host0(Usb, Ready), Active
Resetting left right link! Unidirectional
Uart: didn't receive ack 48984, resending
------ pm update: someone awake 1

uhk80:right$
uhk80:right$
uhk80:right$

// put pc to sleep.
// note that keyboard_app never calls the related Connections_SetState to report that the connection host ceases to be available.
// As a result, power manager still sees a connected host and concludes that uhk should not enter sleep mode.

uhk80:right$ =============
L2 Suspend
------ pm update: someone awake 1

uhk80:right$
uhk80:right$
uhk80:right$
uhk80:right$

// woke up - no logs at all;
// note that usb power handler doesn't report waking up from suspend
// keyboard_app still thinks it is connected, and doesn't report anything either

// tap some keys
// Shows that we have given reports over to c2usb, but they weren't delivered

uhk80:right$
uhk80:right$ Giving report to c2usb usb hid!
Giving report to c2usb usb hid!
Giving report to c2usb usb hid!

```

reesz, 2025-03-21:
Adding some more data points:

Initially I was encountering this issue: https://github.com/UltimateHackingKeyboard/firmware/issues/1144

After flashing firmware `12.4.0 #message-pool-hunt-v1.4` I have not encountered this error message so far.
The connection works correctly wired or via direct BT connection (no dongle) as long as the PC runs.

Once I shutdown my PC (Windows 11) over night, the device does not connect in any way the next day.
If the last connection was Bluetooth it will reconnect every 500ms or so & immediately disconnects again, until I remove the BT connection on Windows & repair.
If the last connection was wired (not unplugged over night) or I try to connect wired after the BT scenario above, the agent will not detect the keyboard until reset.
Also showing the "USB device not recognized" message eventually. 

After pressing reset & reconnecting, the agent will show that the right side seems to be broken offering the "agent can fix it", which it does. Allowing me to reconnect wired afterwards.

Have not been able to extract logs from the agent for this.

benedekkupper, 2025-03-23:
Please give https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14018918080/artifacts/2803151383 a try.

mondalaci, 2025-03-24:
Works for me! Put my Windows laptop to sleep during the night, and when waking it up, instead of the usual OLED Disconnected status it just worked.

Any other feedback, guys? I want to merge this, but I'm waiting for your findings.

kareltucek, 2025-03-24:
While restarting the system:

- grub lags terribly when uhk80 is connected
- right half cycles in a crash loop: 

```
I: No ID address. App must call settings_load()
Settings: Found peer 'left' with address 5922e50268f7
I: Identity: CA:06:14:C4:D3:8F (random)
I: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
I: LMP: version 6.0 (0x0e) subver 0x104e
Reading hardware config
Reading user config
Applying user config
Flashed User Config version: 9.0.0 (native version: 9.0.0., at 25749d7 / 0e2877b
111b4c0c074c347b3afd92b2e)
Clearing bonds
 - Keeping a known bond.
 - Keeping a known bond.
 - Keeping a known bond.
 - Keeping a known bond.
 - Keeping a known bond.
Resetting left right link! Unidirectional
User config applied
Bt: NUS Server module initialized.
Scan module initialized
Bt: NUS Client module initialized
Bt: Starting bluetooth services.
Bt: Starting advertising and scanning, try 0!
Bt: Adv: HID "and NUS" advertising successfully started
USB config changing to Hid_NoGamepad
I: CTRL EP setup: 00 05 0022 0
I: CTRL EP setup: 80 06 0100 8
I: CTRL EP setup: 80 06 0100 8
E: Failed to allocate net_buf 920
E: ctrl_buf alloc fail, stall
ASSERTION FAIL @ WEST_TOPDIR/zephyr/lib/net_buf/buf.c:460
E: r0/a1:  0x00000004  r1/a2:  0x000001cc  r2/a3:  0x20000000
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x00034e1d
E:  xpsr:  0x01000000
E: r4/v1:  0x00000000  r5/v2:  0x00000000  r6/v3:  0x00000000
E: r7/v4:  0x00000000  r8/v5:  0x00000000  r9/v6:  0x00000000
E: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x200357e8I: No ID address. App must call settings_load()
```

Not sure if this is related to c2usb though.

kareltucek, 2025-03-24:
Uhk still works after a sleep-wake cycle (win 11, same as before), producing this log:

```
//win 11, put to sleep:
uhk80:right$ I: CTRL EP setup: 00 03 0001 0

// wakeup again:
uhk80:right$ I: CTRL EP setup: 00 01 0001 0
I: CTRL EP setup: 21 0a 0000 0
I: CTRL EP setup: 21 0a 0000 0
I: CTRL EP setup: 21 09 0201 2
I: CTRL EP setup: 21 0a 0000 0
I: CTRL EP setup: 21 0a 0000 0
I: CTRL EP setup: a1 01 0303 2
I: CTRL EP setup: 21 09 0303 2
I: CTRL EP setup: a1 01 0303 2
I: CTRL EP setup: 21 09 0303 2
I: CTRL EP setup: 21 09 0201 2

uhk80:right$
```

(I didn't check whether uhk is notified about the sleep correctly. The posted build's commit isn't available in git, so I am not sure what logging is and isn't supposed to be present.)

mondalaci, 2025-03-24:
I'm also affected by the crash loop after reboot, which stops after the boot process lefts grub.

benedekkupper, 2025-03-25:
Let's move this discussion over to the PR #1189