UltimateHackingKeyboard/firmware #1152: Mouse not moving diagonally in slow mode on Linux

Nicd, 2025-02-27:
The 12-something firmware has done something to the mouse key system, making the experience worse than it was in I think 11-something. I can deal with the new scrolling behaviour, but slow mode mouse movement is broken. I've taken an example video of this:

https://github.com/user-attachments/assets/63635049-e76f-4631-924d-d3ee4d10e206

As you can see (especially at the end), the cursor is not moving in a 45 degree direction when pressing two keys simultaneously. Instead, it seems to act as if it picked a random percentage of the other movement to apply from 0 to 100 %. For example, if I'm moving left and I then also press up, it might continue going left, change completely to going up, or choose any diagonal direction in between. I would expect it to go in a 45 degree direction.

It seems that this does not happen when moving the mouse at normal or extra speed.

I'm unable to share my configuration at this time, but my mouse keys are nothing special. I don't have advanced macros that would alter this. I'm using firmware 12.4.0 on a UHK60, my system is an Ubuntu Linux system with GNOME.

[image]

mondalaci, 2025-02-27:
I know there's something wrong with the mouse speed because I experienced faster speeds when using it via BLE, and sometimes it was unusually fast, but I can't reproduce it consistently. Can you?

Possibly related issue: #1131

mondalaci, 2025-02-27:
Please let us know the exact firmware version that introduced this issue.

Nicd, 2025-02-27:
I don't know right now, my keyboard is already on 12.4.0 but it must've been 11-something before that. I have another UHK that I could test upgrading release by release with.

Nicd, 2025-02-28:
I upgraded my other UHK60 v1 (both are v1). This behaviour starts on firmware 12.0.0.

mondalaci, 2025-02-28:
I can't reproduce this issue even with your mouse speed settings. Please share your configuration.

Nicd, 2025-03-01:
> mondalaci

I’ll be unable to share a configuration in the next week as I’m on vacation right now and my keyboards are at work. I’ll get back to you on that later, I’ll try to make a minimal configuration that exhibits the issue.

Just a thought I had, does this new firmware require more transfer capability from the middle cables? I have my own custom cables that have worked fine so far (since 2018), but if the firmware has been rewritten to transfer more data over them / require tighter timings, maybe they’re not up to spec anymore. I don’t know how the firmware works so it’s just a thought.

mondalaci, 2025-03-01:
The new firmware doesn't require more transfer capability.

GlowingUmbreon, 2025-03-04:
I am also experiencing the same issue on my UHK 60 v2, in addition to this problem i am also experiencing stutters if you try to scroll and move the mouse at the same time and activating a macro while the mouse is moving is inconsistent it does **feel** like there is not enough bandwidth. This also began occuring on the version 12 update for me. I have already tried resetting the config on the keyboard, plugging it into a different laptop with no other peripherals plugged into it and decreasing the baud rate through a macro however none of this seems to have effected the problem. Is there any logs or anything I can provide to help with this problem? This is happening on windows

mondalaci, 2025-03-07:
I still can't reproduce this issue.

@Nicd @GlowingUmbreon Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/13703847914 build fix the issue for you, guys?

@kareltucek Which UHK models are affected by this issue?

kareltucek, 2025-03-07:
Uhk60.

Nicd, 2025-03-10:
Can confirm, this fixes the mouse issues for me too on UHK60 v1. I can't reproduce GlowingUmbreon's scrolling issue.

Thanks! 💜