UltimateHackingKeyboard/firmware #1155: Slow scrolling after wake up

mondalaci, 2025-02-28:
Scrolling is extremely slow after resuming Linux from sleep. Reproduced with kernel 6.8.0 and 6.13.3

When looking at the output of `git log`, 70b297d1 works fine and seemingly 33d8d207 introduced the issue, however `git log --oneline 70b297d1..33d8d207` reveals:

```
33d8d207 Merge remote-tracking branch 'origin/master' into high-res-scrolling
9f1f69c5 Fix high-ress scrolling multiplier computations.
95e7a30f Merge remote-tracking branch 'origin/master' into high-res-scrolling
c621251a (HEAD) add high resolution scrolling to UHK-80, switch to standard defined implementation
```

c621251a contains the issue, so I think it's the culprit.

This regression affects the UHK 60 and UHK 80.

@benedekkupper Can you fix this issue?

mondalaci, 2025-03-01:
There's an important detail I realized and want to highlight:
- For c621251a, the issue can be experienced right after flashing it and also after a sleep-wake cycle.
- For 33d8d207, the issue cannot be experienced right after flashing it, only after a sleep-wake cycle.

kareltucek, 2025-03-06:
I can't reproduce this with neither Windows, nor linux.

kareltucek, 2025-03-10:
Anyways @benedekkupper, it seems like after a sleep-wake cycle, with some hosts the host thinks the device uses high-res scrolling (120 mulltiplier), while UHKs (both models) use normal low res scrolling (1 multiplier).

(If I read the code correctly, it means that (with some hosts), uhk receives the kUSB_DeviceEventSetConfiguration event, but does not receive the kUSB_DeviceHidEventRequestReportBuffer afterwards.)

bhaeussermann, 2025-03-11:
I'm experiencing this issue on Windows each time I boot up after a shut-down. As soon as I do a sleep-wake the scrolling returns to normal, or when I unplug and plug the UHK it also returns to normal.

I'm using firmware version 12.3.5 on UHK 60.

dEEkAy2k9, 2025-03-12:
I am using my UHKv2 (UHK 60) on a Thinkpad laptop and on a desktop PC. The keyboard is directly connected to the laptop via the OG usb-c to usb-c cable. After booting up the laptop, scrolling is super slow, both with the left thumb cluster (small ball) and via modifier + trackpoint/trackball (haven't tested touch module).

After disconnecting and reconnecting the usb-c cable, everything is fine.

This is on Windows 11 24H2 Build 26100.3194.

The same behaviour happens on the desktop PC, Windows 11 24H2 too but connected either through a usb switch and/or with a usb-c to usb-a adapter.

UHK Firmware is on 12.3.3

[image]

mondalaci, 2025-03-12:
Reconnecting USB fixes the issue for me, too.

This issue is not module-related, but USB-related, and affects mouse keys and all modules.

timnash, 2025-04-01:
I have also been experiencing this with the UHK 60 v2 on Windows 11 with firmware 13.0.0 (and on previous firmware but I don't remember the exact versions). Mine is connected to my monitor, which is connected to my computer via Thunderbolt.

Unplugging the UHK and re-plugging back in fixes the issue.

dEEkAy2k9, 2025-04-03:
I noticed something.
I have been using my uhk60 v2 through my work laptop's hub and whenever that hub powers down and back up, the issue comes back. Unplugging and replugging fixes it.

Connecting the keyboard directly to the laptop only gives me that issue once after first boot and is gone after replugging the usb cable. Kinda strange.

Maybe something with power distribution causes it as powering down and up again the hub reproduces it.

rasmusloje, 2025-04-18:
Can reproduce consistently when starting up PC after power has been off the outlet (controlled via remote plugs). I'm on firmware 13.0.1.

Reconnecting USB or restarting PC fixes the issue.

mondalaci, 2025-05-16:
It's not the same issue. We'll fix this one separately.

dEEkAy2k9, 2025-06-05:
> Seeing this on a UHK 80 v1 on Windows and keyboard firmware 14.0.1. Issue seems to occur every 20-30minutes or after locking/unlocking my laptop where the laptop is USB-C docked to a Dell monitor dock and the keyboard is plugged into the monitor.
> 
> As a workaround (using a tip [from here ](https://forum.ultimatehackingkeyboard.com/t/mouse-scrolling-slow-at-pc-startup/2306)) I've bound one of my Fn layer mappings to a `reboot` keyboard command so I don't have to keep unplugging/replugging the keyboard.

The `reboot` command is useful, thanks for that.

Any idea on what actually causes this issue? UHKv60 here.

kareltucek, 2025-06-05:
> Any idea on what actually causes this issue? UHKv60 here.

Probably it is related to high res scrolling and usb quirks of various OSes.

@benedekkupper could you maybe look into it?

oz123, 2025-08-14:
> Reconnecting USB fixes the issue for me, too.
> 
> This issue is not module-related, but USB-related, and affects mouse keys and all modules.

I am experiencing similar issues with :

UHK (UHK 60 v2)

 Right keyboard half firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Left keyboard half firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Key cluster firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Trackball firmware: UltimateHackingKeyboard/firmware: 15.0.1

Kernel 6.12.x-6.14.0 (on gentoo).

My left  Key cluster roll isn't vertically scrolling after suspend. Unplugging the USB cable of the  keyboard and reconnecting it solves the issue.

(Side note, since I never knew if it is supposed to be the case: cat that little trackball also do horizontall scrolling?)

mondalaci, 2025-08-14:
@oz123 Can you reproduce this issue with firmware 15.1.0?

It should be possible to scroll vertically and horizontally by using the key cluster's mini trackball.

kareltucek, 2025-08-14:
@benedekkupper I will try to retest tomorrow morning. Afaik, Laszlo isn't affected.

benedekkupper, 2025-11-15:
I need everybody's help to move this issue forward. This problem is caused by a bug in the Linux kernel, so I ask all of you to report it to your distro's maintainers. I have submitted a MR over here: https://lore.kernel.org/linux-input/20251007203544.9963-1-kupper.benedek@gmail.com/T/#u , but it's unlikely to be picked up anytime soon, without user demand. So don't forget to reference it in your bug reports.

As an intermediate workaround, you can check out this PR in agent: https://github.com/UltimateHackingKeyboard/agent/pull/2744

bhaeussermann, 2025-11-16:
> Just reported https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2131607
> 
> If anyone else reports it, please paste the links here.
> 
> We can ask the 10K+ UHK mailing list subscribers to report/upvote this bug in one of our upcoming newsletters.

Though as per the comments above, many users (including myself) are experiencing this issue on Windows. Is that a separate issue?

benedekkupper, 2025-11-16:
> Though as per the comments above, many users (including myself) are experiencing this issue on Windows. Is that a separate issue?

That does seem like a separate issue. From what I gathered here, it is exclusively in the scenario where the keyboard is connected through a monitor, or other hub with external power supply, is that correct? Please follow up with a separate ticket, describing as much of the circumstances as possible.

jscheidtmann, 2025-12-15:
Windows users please add to #1406!

rightaditya, 2026-02-24:
Yep, 6.19 fixes it for me (CachyOS)! Congrats @benedekkupper!!