UltimateHackingKeyboard/firmware #1175: Ignore `reboot` in macros starting with `$` and throw a status buffer warning

mondalaci, 2025-03-10:


mhantsch, 2025-03-11:
I am not totally sure how the logic of the macro engine works, but assuming there is a scheduler that has slots for each running macro, here is what I would try:

When starting to run a macro that should have a `reboot` exception, set an internal flag in that scheduler slot. Ignore `reboot` commands as long as that flag is set. The flag would carry over to other macros that are `exec`'d. Maybe even to those that are `fork`ed. The only macros where `reboot` would work are those that are started from a keypress, which would not have that flag set.

This will still allow to program a key that reboots.

Or phrase the flag in reverse: only execute `reboot` in macros that were started from a keypress and it's `exec`'d or `fork`ed descendants.

I can only think of one questionably useful use case which would not be allowed: something like "auto-reboot the UHK every 15 minutes".

kareltucek, 2025-03-11:
I still don't see how anyone could run into this by accident, and don't want to increase codebase complexity over it.

Let's return to this if this actually becomes a problem.

closed 2025-03-11 by kareltucek