UltimateHackingKeyboard/firmware #1176: UHK60v2: LED Segment Display Remains On After Computer Turned Off

nlordell, 2025-03-11:
A similar issue was first described in https://github.com/UltimateHackingKeyboard/firmware/issues/1123, however that issue has since been closed.

In my setup I have the keyboard connected to a powered USB hub, which is connected to my computer. When turning off the computer, the LED segment display (which by default shows that current keymap) **remains on**, despite key backlight LEDs being correctly powered off.

This appears to be related to v12.x firware versions (as I did not observe it until flashing v12.3.3 recently and was previously on v11.2.0). Note that I am unable to downgrade to a v11.x firmware, as the Agent v4 cannot detect the keyboard and Agent v5 cannot flash v11 firmware.

As a workaround, I currently set the LED display brightness to 0, which is not ideal. Note that setting a LED display timeout does not seem to have an effect in this state (the LED segment display remains on even after the timeout).

kareltucek, 2025-03-11:
Well, it should be possible to downgrade, although you may need to drop into bootloader manually: https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half (with the 4.x Agent).

nlordell, 2025-03-12:
Thanks, I will do that if my workaround gets too annoying (i.e. I start missing the segment display too much).

I also wanted to report the issue to make sure it is documented somewhere.

mondalaci, 2025-03-12:
I can't reproduce this issue with firmware 12.4.0, so please flash it and retry.

When I put my Linux PC to sleep, the key backlighting and the LED display of my UHK 60 go out.

If you can reproduce it with 12.4.0, please let me know your OS and attach your configuration.

nlordell, 2025-03-13:
> I can't reproduce this issue with firmware 12.4.0, so please flash it and retry.

The issue persists.

> When I put my Linux PC to sleep, the key backlighting and the LED display of my UHK 60 go out.

Some additional clarification:

* As mentioned above, the keyboard is **not** connected to my computer directly, but instead over a USB hub that remains powered when the computer turns off. Below is a diagram of my setup (my UHK is connected to the USB hub from my Apple Studio Display which itself is connected to a Mac Mini).

[image]

> please let me know your OS

This happens when connected to the USB hub from my Apple Studio Display _regardless of the computer connected to the display_ (I tried it with my laptop running Fedora 42, and my Mac mini running macOS 15 connected to the display, as well as just connecting it to the display with no computer connected to it all all - the issue is always observed). Here is a picture of my usual setup - note that my computer is turned off:

[image]

This issue also seems to happen regardless of the USB hub being used, the key being that is **powered**. For example, I have an Intel NUC (running Fedora 42, but I don't think that matters) - if I connect it to the right USB (the yellow one [which provides power even when the computer is off](https://www.corsair.com/us/en/explorer/diy-builder/storage/usb-port-colors-explained/)), and power off the computer, the LED stays on:

[image]

However, connected to the left USB (which only has power when the computer is turned on), then it does not happen:

[image]

In fact the easiest way to reproduce this, is just connect the UHK to a laptop charger, you will see the LED turn on:

[image]

This behaviour seems to have started in v12.x onwards (so I did not observe this in v11.2.0 which I was running for a while).

> and attach your configuration

I "Reset Device Configuration" in UHK to make sure it wasn't configuration specific before reporting, so I am using the default configuration for Agent 5.1.0.

nlordell, 2025-03-13:
Also, just to clarify: at least for me, this is a fairly minor issue (especially since I've found a way to work around it for now), I just wanted to report it so that it is documented as I imagine it is related to, from the 12.0.0 release notes:

> The UHK 60 v1 and UHK 60 v2 may experience some regressions due to major refactors, but this firmware should be well-suited for everyday use, as we are not aware of any major issues.

mondalaci, 2025-03-14:
@kareltucek Please try to reproduce this issue and let us know whether you can.

nlordell, 2025-07-06:
I've recently upgraded from firmware v12.* (don't remember the exact version) to v14.2.0 and noticed that the LED backlight timeout is no longer respected when the device is connected to a powered USB hub.

Again, the easiest way to reproduce is to plug the keyboard to a USB charger. It used to be that the LED backlight would turn on, and after the LED backlight timeout, it would turn off by itself, but this is no longer the case and the LED backlight stays on indefinitely regardless of the timeout setting (so, the problem I was having with LED segment display now also applies to the LED backlight).

I'm currently working around this by manually entering sleep mode.

mondalaci, 2025-07-11:
@kareltucek Is this a USB stack related issue?

kareltucek, 2025-07-11:
Well, led timeouts should be working no matter what usb reports 🤨.

kareltucek, 2025-09-24:
Should be fixed in https://github.com/UltimateHackingKeyboard/firmware/pull/1362 . 

Please give a try to the firmware from https://github.com/UltimateHackingKeyboard/firmware/actions/runs/17977250372

jpyllman, 2025-09-25:
Does UHK get any HID info on USB if it gets disconnected from any HID device and only getting power???

nlordell, 2025-09-27:
> Should be fixed in [#1362](https://github.com/UltimateHackingKeyboard/firmware/pull/1362) .

Cool thanks! Just out of curiosity, does this fix apply to _both_ the LED backlight **and** segment display? They were both an issue for me (just trying to avoid having to open up the keyboard again in order to install the older v11 firmware again).

PR #1362 (2025-10-06): Fix UHK60 backlight sleep mode.
Closes #1176 

Reproduction steps:
- set backlight fade timeout on a UHK60 to say 4 seconds
- connect uhk60 into a dumb usb charger (the older, the better)
- observe that after four seconds it doesn't turn its backlight of
- in case it does, tap some key (a one that has a scancode mapped)
- again observe that after four seconds it doesn't turn its backlight of
closed 2025-10-06 by kareltucek