UltimateHackingKeyboard/firmware #1186: Mouse speedup and jumps

mondalaci, 2025-03-18:
There's a bug that makes my mouse cursor faster and jumpy. This issue affects both mouse keys and modules. It usually happens after reflashing firwares. I think I usually flash the left firmware and then the right and then it happens, but I can't reproduce it consistently. As for connectivity, I use USB and the bridge cable.

I can easily notice when this bug hits, as the initial mouse speed gets noticeably faster, and it becomes hard to hit targets precisely with the mouse pointer. Also, the mouse pointer teleports every few seconds when moving the mouse.

@kareltucek Please provide a Zephyr command that dumps mouse speed multipliers and any additional information that you find relevant to investigate this. I'd merge it to master because I test many branches, and it'd be easier to debug the issue whenever it happens.

kareltucek, 2025-04-02:
Module mouse controller has no speed multipliers common with mouse keys controller.

Please try to switch to a dongle next time it happens, and observe whether it fixes the issue.

mondalaci, 2025-04-02:
It's not stuck acceleration because it's not that fast, and mouse jumping is a frequent side effect. When I encounter this bug, I'll switch to the dongle and try `mouseMultipliers`.