UltimateHackingKeyboard/firmware #1188: Trackpoint/mouse layer/key cluster scroll stop functioning intermittently

bitmapdrone, 2025-03-20:
Running firmware uhk-firmware-12.4.0-message-pool-hunt-v1.4, the trackpoint module and mouse layer intermittently do not function.  Trackpoint, scroll, and mouse clicks all fail to register.

I installed uhk-firmware-12.4.0-message-pool-hunt-v1.4 out of the box when I received my keyboard since it was showing the message pool error out of the box.

I just started using the keyboard as a daily driver this morning, and haven't had a chance to test/troubleshoot thoroughly.

Currently connected to M4 Max MacBook Pro with dongle, but right half USB is also connected for Agent.

Keyboard functionality seems normal.

kareltucek, 2025-03-20:
Hmm, is you dongle flashed to the same firmware too?

bitmapdrone, 2025-03-20:
It was not - just re-ran the flashing process with uhk-firmware-12.4.0-message-pool-hunt-v1.4; will report back in this issue later today - the issue was pretty consistently replicatable every few minutes.

bitmapdrone, 2025-03-21:
Looks like the dongle firmeware version was the issue here.  I'll go ahead and close this issue.

closed 2025-03-21 by bitmapdrone